TheThingsArchive/ttn #37: Rewrite tests to make consistent use of Go Convey

johanstokking, 2016-02-12:


KtorZ, 2016-02-15:
Is it really necessary ? 
I mean, that's about rewriting 60% of the current code base for no real benefits (I see here more complications than actual outcomes)
Go already provides in itself nice test mechanisms and I don't really see the need for a framework like Convey (unless maybe for the visualization / stats parts). 

Although I was avoiding it on purpose, we can write upcoming tests with Convey. Yet I strongly go against rewriting the existing tests.

htdvisser, 2016-02-18:
I don't think this is high priority at the moment. However, from now on we should be consistent in how we write tests. Let's list al arguments for using or not using Convey, have a discussion about this and make a decision about how we will write our tests.

htdvisser, 2016-04-05:
Goconvey is no longer used in `github.com/TheThingsNetwork/...`

closed 2016-04-05 by htdvisser