TheThingsArchive/ttn #39: Finalize downlink

KtorZ, 2016-02-16:
Considering Class A devices, downlink transmissions are merely response to an uplink. 
We'll go for something simple at first:

Applications can publish packets to a queue with a given expiration date / TTL. 
When a downlink response frame is available, a packet is pulled from the queue until a valid one is grabbed. 
The packet is wrapped with the response, and a similar mechanism is set up for the Network Controller and MAC commands. 

Then, the router has to determine the best gateway and window to which forward the answer. For the first sprint, a gateway will be picked at random and we'll consider making a cleverer choice in the next sprint.

tftelkamp, 2016-02-18:
Gateway selection: highest SNR value 
Window selection: SF7-SF8 in RX1, SF9-SF12 in RX2 (at SF9)

KtorZ, 2016-02-29:
Done cde75155a20e132991af681a3c6d27de8b6251a3.

closed 2016-02-29 by KtorZ

jtuki, 2016-03-02:
It seems that https://github.com/TheThingsNetwork/ttn/commit/cde75155a20e132991af681a3c6d27de8b6251a3 is a refactor commit?

KtorZ, 2016-03-02:
That commit is the merge commit of a refactor which also implements the feature (without selection between two windows nevertheless; this is coming in the next few days). 
Because there's several commits linked to that specific issue, I pin-pointed the one that actually brought it into the branch instead of giving the exact list (which would have been a bit too harassing :) )

htdvisser, 2016-03-02:
I think we should try to be more consistent in using feature branches, @KtorZ. Otherwise we could just as well rename `develop` to `trunk` and do all development there ;)