TheThingsArchive/ttn #42: Test and Handle OTAA

KtorZ, 2016-02-16:
Last step of the the first Sprint, the implementation of the Over-The-Air-Activation process. This will require to strengthen the responsibility of the Broker and its Network Controller. 
Nothing is settled yet about the way a handler is supposed to confirm a registration to an application.

cambierr, 2016-03-16:
Hi @KtorZ !
I was wondering if anything has been moving in this issue ? We are developing a custom handler for our special use-case, but waiting for the protocol definitions of the join sequence :)

Thanks!

KtorZ, 2016-03-16:
Hey, I am on it, this week. 

What you can expect for the protocol between broker and handler is [gRPC](http://www.grpc.io/) and [protocol buffers](https://developers.google.com/protocol-buffers/) such that everything is language independent. 

For the OTAA, you can expect the following message:

```proto
message JoinRequest {
    bytes       AppEUI          = 1; // 8-bytes long
    bytes       DevEUI          = 2; // 8-bytes long
    bytes       DevNonce        = 3; // 2-bytes long
    bytes       AppNonce        = 4; // 3-bytes long
}
```

This is not completely settled already however, it will be really close to that. I hope it helps.

cambierr, 2016-03-16:
Yep, that's exactly what I wanted to know! thank you for fast reaction!

commit 2dbd8890 (2016-03-21): Merge branch 'feature/otaa' into develop. Close #42.
closed 2016-03-21 by KtorZ