TheThingsArchive/ttn #46: HTTP adapters should ignore "broken" servers

htdvisser, 2016-02-18:
Temporarily disabling broadcasts/sends to servers that return errors (such as `net/http: request canceled (Client.Timeout exceeded while awaiting headers)` or `dial tcp host:port: getsockopt: connection refused`). Maybe add something like exponential backoff.

KtorZ, 2016-02-29:
Request are ignored for the moment. The non-responding recipients remain as potential recipient though. It could be nice indeed to filter the recipients list dynamically. Not a priority, but still a possible optimization.

KtorZ, 2016-03-09:
Done 5054e214828e399e545ab646a7d843e2ba5222e8

htdvisser, 2016-03-09:
Nope. Not done.

htdvisser, 2016-03-25:
For now The Things Network Foundation hosts the servers, which means that we have control over them. As soon as we open the network up to third parties, we need this functionality.

closed 2016-10-12 by htdvisser