TheThingsArchive/ttn #50: Calculate Duty Cycle Information for Each Gateway

htdvisser, 2016-02-19:
In most cases a Gateway is connected to **one** Router. This means that the Router should calculate duty cycle information (based on `Datr` and `Size` from `RXPK`)

FRVisser, 2016-02-24:
Is there not a function for this in the package forwarder?

htdvisser, 2016-02-25:
We want to be able to calculate airtime information for each **node**. This will be necessary for monitoring (and enforcing) the fair access policy (the 30s airtime) for nodes.

I agree with @FRVisser that calculating the gateway's duty cycle _should_ be the gateway's responsibility, and I think we should include this functionality in the firmware for The Things Gateway. Unfortunately the default packet forwarder doesn't send this information to the server. We want to be able to calculate (or at least approximate) the duty cycle for these gateways.

KtorZ, 2016-03-09:
Done 1a0f3ce509854f4fa4c3f297d3d5eb262d773400

closed 2016-03-09 by KtorZ