TheThingsArchive/ttn #58: Fix Paho library

KtorZ, 2016-02-25:
We use the [paho](https://github.com/KtorZ/paho.mqtt.golang/) library as an go `MQTT` client which works nicely for the moment. 

The library defines a `ClientInt` as an interface for the client, however, they also declares a public type `Client` which implements that interface and use that type within the latter interface. 

The `ClientInt` is defined with:

```go
type ClientInt interface {
    ...
    Subscribe(string, byte, MessageHandler) Token
    SubscribeMultiple(map[string]byte, MessageHandler) Token
}
```

where MessageHandler is

```go
type MessageHandler func(*Client, Message)
```

And this... well.. makes the interface almost useless because we cannot build mock component properly anymore. You're stuck with the Client implementation. 
For now, I've made a small workaround but it could be nice to make a PR to the paho lib in a while to fix that and:

- Use the `ClientInt` in the `MessageHandler` type (and even use an interface for that type)
- Remove the `Client` struct from the exported entities
- Rename the `ClientInt` in `Client`. 

I don't know yet if the above changes are possible. They can make the whole a bit better to use though.

KtorZ, 2016-02-25:
```go
type Token interface {
	Wait() bool
	WaitTimeout(time.Duration) bool
	flowComplete()
	Error() error
}
```

:'( 

Have to do something with that `flowComplete()`

KtorZ, 2016-03-14:
https://github.com/eclipse/paho.mqtt.golang/pull/34

https://github.com/eclipse/paho.mqtt.golang/pull/33

htdvisser, 2016-04-05:
We're not experiencing any problems with Paho in our staging environment. Closing this issue.

closed 2016-04-05 by htdvisser