TheThingsArchive/ttn #64: Enhance Code Quality

KtorZ, 2016-03-02:
Make use of [golint](https://github.com/golang/lint) to ensure the code written is completely conform to the go best practices. 

```bash
go lint ./...
```

Not a priority though, but a nice thing to do at the end to polish before the release.

htdvisser, 2016-03-02:
If you use the [go-plus](https://atom.io/packages/go-plus) package for Atom these things are shown right in your editor!

KtorZ, 2016-03-02:
I don't use atom. I would rather use a text editor :)

Showing them is a thing, applying them and making sure they're applied is another. I've started already to make some changes, at some points, we'll be totally compliant.

turiphro, 2016-03-03:
Not sure if you're using this already, but the Vim Go plugin automatically formats your code with gofmt on every save, fixing most issues already:

https://github.com/fatih/vim-go

You can add golint highlighting with:
set rtp+=$GOPATH/src/github.com/golang/lint/misc/vim
autocmd BufWritePost,FileWritePost *.go execute 'Lint' | cwindow

Works quite smoothly.

KtorZ, 2016-03-03:
Hey @turiphro, thanks for the tips. I am already using all of this though :) 
Vim by itself is a bit rough ^.^, got a combination of golint + go fmt + syntactic (auto-completion and package introspection).

closed 2016-03-18 by KtorZ