TheThingsArchive/ttn #69: Improve internal communication protocol

johanstokking, 2016-03-08:
Replace existing custom binary protocol over HTTP

Consider a faster binary serialization, standardized across components preferably using proven technologies (e.g. Google Protocol Buffers, gRPC over TCP or HTTP/2)

htdvisser, 2016-03-09:
I think we should define **one** synchronous protocol and **one** asynchronous protocol that we will use in the entire stack.

## Current situation

Let's look at what the existing communication looks like. The largest part of the communication is already request/response:

For uplink:
- [x] Gateway-Router: `PUSH_DATA`/`PUSH_ACK`
- [x] Router-Broker: `HTTP` req/res
- [x] Broker-Handler: `HTTP` req/res
- [ ] Handler-Application: `MQTT` publish (no response)

For downlink:
- [ ] Application-Handler: `MQTT` publish (async, so no immediate response, although the application usually wants an acknowledgement)
- [x] Handler-Broker: `HTTP` req/res
- [x] Broker-Router: `HTTP` req/res
- [ ] Router-Gateway: `PULL_RESP` (no ack)

For ABP registrations:
- [ ] Application-Handler: `MQTT` publish (no response, although the application usually wants an acknowledgement)
- [x] Handler-Broker: `HTTP` req/res

For OTAA: 
- First part is the same as uplink
- [x] Handler-Application: `MQTT` publish (req)
- [x] Application-Handler: `MQTT` publish (res)
- Last part is the same as downlink

### The Transport Protocols

`MQTT` is a great, lightweight protocol for asynchronous communication. It's especially well suited for many-to-many or one-to-many communication.

`HTTP` is well suited for one-to-one synchronous communication. `HTTP/2` has very little overhead, so it's also lightweight.

### The Payload

In most components, we send non-compressed `JSON` payloads (over `HTTP 1.1`), which is not ideal. We also send binary data that is encoded/decoded with custom marshalers/unmarshalers, which is probably also not exactly what we want. We also send binary data (over `MQTT`) to the application and back, which is also not what we want. We clearly have to find a suitable alternative.

A smaller alternative to JSON is [`msgpack`](http://msgpack.org/), but we're defining an internal protocol, which means that we can specify the exact data structure that has to be sent from one component to another. As a result, we don't need (want) the flexibility of `JSON` or `msgpack`.

I think it's a good idea to use [protocol buffers](https://developers.google.com/protocol-buffers/), which are a language-agnostic way to define and encode/decode messages between components.

It's easy to send protocol buffers over any transport protocol (probably `HTTP/2` and `MQTT`).

### `gorilla/rpc` and `gRPC`

[`gorilla/rpc`](http://www.gorillatoolkit.org/pkg/rpc) supports multiple codecs, so it might be easy to migrate from what we have now. 

[`gRPC`](http://www.grpc.io/) works nice with protocol buffers, works with multiple languages, is lightweight and seems to be quite well supported. Let's have a look at this and see if it fits our requirements.

KtorZ, 2016-03-11:
I had a deeper look into protocol buffers. And, there's something that bother me. They don't support fixed-size array. We're manipulating a lot of them (AES128Key, DevEUI, DevAddr, AppEUI, GatewayID, DevNonce ...).
On the other hand, I found this [msgp](https://github.com/tinylib/msgp) which enables fast and simple marshaler / unmarshaler code generation (even faster than protobuf unless we go for [gogoprotobuf](https://gogo.github.io/) which seems the fastest and still reliable implementation). 

  | Protocol Buffers | Msgpack & `msgp`
-----|----------------|--------------------------
Pros | Language-neutral, Getters & Setters generation, Blazingly fast (with `gogoprotobuf`) | Easy-to-implement (go annotation), More flexible (any go types and structs), Blazingly fast as well, Free tests & benchmark generation 
Cons | Heavier to setup (dedicated language in dedicated files), external c++ dependency, Limitated amount of types (no fixed-size arrays, no maps ... ) | Less interoperable, No Getter & Setter generated

I find Protocol Buffers really elegant, and, we could work well with them if we can get rid of every fixed-sized arrays, and replace them with slice of bytes. That won't change a lot for us (and could maybe make the code prettier in some way, we won't have to 'cast' types from `lorawan` libraries everywhere), but we loose static type verification doing that. Nevertheless, if we only allow messages to be created from marshaling methods, then, we can ensure the correctness of a message by strongly typed the message itself (even more than what we're currently doing).
Plus, @johanstokking mentioned it a couple of times already; Should we want to raise a bit the abstraction about packet identifiers and keys to also support bluetooth and whatever-we-will-want-to-support, using plain slices of byte as identifier would go in that sense (especially when go does not support genericity and won't support it in further versions). Most of the libraries (all ?) we're using consider slices of bytes as identifiers or key. The type verification could be given by the type of the interface itself, so, calling for instance `(p RouterPacket) func packet.DevAddr() []byte` on a router packet, one can assume with 100% certitude that the result will be a slice of 4 bytes *because of the nature of the packet itself*

Using MsgPack nevertheless can offer us a clearer, non-implicit typing. I like both solution. Protocol buffers are nice because they enable a huge interoperability between machines.. However, do we really need that ?

closed 2016-03-18 by KtorZ

q2dg, 2018-06-15:
Why this issue was closed? Is there any conclusion? Thanks