TheThingsArchive/ttn #70: Router Discovery

htdvisser, 2016-03-09:
I'm opening this issue to start discussing the way a gateway will find a TTN router to associate with.

Initial options to start the discussion:

* **Configuration**: Configure an IP address and port in the gateway (hostname if DNS is supported). 
    For DNS: how to find the best router?  
    Use region in hostname? `eu-west.***`/`us-east.***`  
    Random/Round-Robin is probably not the best solution.  
    Something like Amazon Route 53’s Geo DNS, maybe?
    Router load/performance is also important
* **Discovery Service**: Gateway contacts a TTN discovery service that returns a number of good candidates, gateway decides which is the best.  
    Good way to authenticate gateways with TTN
    Include router load/performance in decision?
    GeoIP based or should the gateway measure RTT?

Beanow, 2016-05-24:
To do _true_ decentralization you will need:

1. A bootstrapping process for new gateways and routers.
2. Every router and every gateway should implement the full blown peer-to-peer discovery spec.

The first part is not too difficult. You can build DNS based lists of recently observed live routers' IP addresses. You can make REST API's. I've seen IRC channels being abused for it even. Or you could give a typing monkey some bananas and have him upload a JSON file to some webserver. Doesn't really matter too much how these first connections are shared.

The second part is where the pain and the importance lies I think. Every gateway _should_ keep their own list of routers they've successfully connected with before, so that they eliminate the need for the centralized bootstrapping service as quickly as possible. They also _should_ periodically ask routers for a list of more routers and try to connect with them to keep a healthy number of potential routers to connect with should a router go down.

That means routers should do a similar thing. Routers should connect amongst themselves using the same bootstrapping process so they can announce themselves as a member of the network. If routers do not implement this, there is no way for a router to attach itself to the network without again depending on the bootstrapping process, which centralizes the network.

An example of this setup would be Bitcoin. A full bitcoin node has a hard-coded list of DNS lookup services where it can find it's first peers, from there on it will keep it's own list on persistent storage and constantly maintain a list of known peers and probe for them at set intervals.

Another can be seen in torrents. Trackers can be seen as the bootstrapping help to speed things up, but if need be you can rely on DHT to eventually give you the same meta information and use peer discovery to find more seeds to download from.

The issue with this is, this involves a large number of network activity, a reasonable uptime so as to not become isolated, some persistent storage and a fair amount of business logic to do. Making it very painful to implement in for example gateway firmware with little to no storage and costly 3G connections. Bitcoin also ran into this problem with mobile wallets which resulted in thin clients and community run servers for those to connect with. Which in terms of risks involved brings you back to square one, but in terms of decentralizing the core infrastructure is still acceptable. Because it is still possible to be fully decentralize if you so wish, but you probably want to do that with your laptop and not your budget android phone.

Beanow, 2016-05-24:
Probably I only covered part of you questions there. Let me give some more concrete opinions.

**Configuration**, the bootstrapping process uses very long lived addresses. The kind where you feel comfortable hard-coding this address into your firmware on a gateway you hope you won't need to touch again in the next 6 months. So I would move as much dynamic stuff as possible away from your configuration. No regions in the hostname, no different IP addresses for different use-cases. You will have to solve this with load balancers, virtual IP addresses and the likes and provide high availability this way.

**Which router is best?** Frankly if you assume autonomous, fat clients trying to leverage TTN we shouldn't care what the best way is right now. The gateways are responsible for this themselves. The top priority for TTN should be to provide appropriate means for gateways to do the evaluation at all. And just because we're nice people, we might share a reference implementation and install it on the kickstarter gateways for you. I would leave room for innovation here and not get hung up on finding the best approach right away.

What I would suggest as the first implementation: try to spread wide, with some soft preferences.

At a fixed interval, say every other hour:

1. Let a gateway connect to a list of routers with a max of 20 to establish: how many gateways is this router connected to, what is the response time of this router?
2. Filter out relatively slow routers, if most are around 50ms, drop the 400ms ones.
3. Sort the remaining routers by number of connections they already have, starting with the least amount.
4. Try associating with the routers in that order until one accepts us.

Should a router become unresponsive before our next alarm to look for good routers, try connecting from the same list out of a cache starting at the top. If that fails do the whole process again.

Example situation, suppose this is the router information we obtained:
```json
[
	{"address": "123.45.67.89:5432",   "gatewayConnections": 3, "responseTime":  "40ms"},
	{"address": "123.45.67.108:12233", "gatewayConnections": 1, "responseTime":  "57ms"},
	{"address": "66.104.67.108:12233", "gatewayConnections": 2, "responseTime":  "30ms"},
	{"address": "4.104.203.108:12233", "gatewayConnections": 1, "responseTime": "269ms"}
]
```

The result would be:
```json

{
	"sortedSuitableRouters": [
		{"address": "123.45.67.108:12233", "gatewayConnections": 1, "responseTime":  "57ms"},
		{"address": "66.104.67.108:12233", "gatewayConnections": 2, "responseTime":  "30ms"},
		{"address": "123.45.67.89:5432",   "gatewayConnections": 3, "responseTime":  "40ms"}
	],
	
	"backupRouters": [
		{"address": "4.104.203.108:12233", "gatewayConnections": 1, "responseTime": "269ms"}
	]
}
```

htdvisser, 2016-10-12:
Router configuration is implemented in the gateway configuration mechanism (part of the account server and The Things Gateway firmware). Load balancing will be done server-side.

closed 2016-10-12 by htdvisser