TheThingsArchive/ttn #71: Broker Discovery

htdvisser, 2016-03-09:
I'm opening this issue to start discussing the way a TTN router will discover TTN brokers.

## Current situation

Upon reception of a packet, a router either
* Broadcasts a packet to all known brokers
* Forwards the packet to brokers that previously accepted the `devAddr` (routing tables)
  * If rejected, then broadcast.

This system obviously isn't very scalable, because:
* Routing tables will grow and become useless
* Broadcasting to a couple 100s brokers is just insane
* There are security issues (many possibilities of DoS)

## Long term

For a long-term solution, we're working with a group of experts that will help us define a perfect solution.

## Short term

For the short term, we need something that is better than the current situation.

Initial option to start the discussion:

* **Registry Service:** Brokers register address range (prefix) with a TTN registry service, so routers only have to broadcast to a smaller number of brokers.

htdvisser, 2016-05-11:
Great discussion. We'll go with the Registry service! 🎉

tftelkamp, 2016-05-11:
With a simple registry we'll get some experience for a long-term scalable solution, like how to deal with changing prefixes, etc.

htdvisser, 2016-07-07:
The central discovery service has been implemented in the `refactor` branch. See #199 for more info.

closed 2016-07-07 by htdvisser