TheThingsArchive/ttn #74: Network Model / Simulation

KtorZ, 2016-03-10:
I was discussing with a friend which pointed out something interesting. We are implementing quite a lot of algorithms, processing and other computations in the network. We're going to monitor the whole thing and try to adapt the network from that.

Wouldn't it be nice to actually create a model of our network and of our algorithm ? (In either R, Python, Julia, Haskell or anything else). The idea could be something relatively simple to get a running simulation of the network packet processing. We get rid of every communication (no udp, mqtt, tcp or whatever), and merely focus on what we are actually doing with data. Then, we can either visualize the propagation, congestion or simply measures from different spot. Data transmission can be simulated with delays (tweakable).

Anyway, it could be really nice to get a simulation of all that stuff. Not for now though, but this is to my mind, a really nice project to work on.

cambierr, 2016-04-15:
I agree to say that it could be useful, but I don't think that writing a separate implementation is a good idea since it would require to keep the "simulator" up-to-date of all the ttn updates.

Instead, what could be done, is a "tracing mode" in ttn. For example with a per-component event-bus where all events could be sent (node sending UP message, gateway receiving UP message, decoding ok, routing to router, router receiving UP message, ...). Then, creating an eventual UI for better comprehension and visibility ?

KtorZ, 2016-04-15:
In such a case, you wouldn't be simulating but monitoring. Those are two different issues and both matter.

cambierr, 2016-04-15:
Not really; if using a node-simulator and running a dedicated instance of the ttn stack, it becomes like a simulator using real algorithms without re-implementation and required maintenance. But with the same result as a pure simulator.

jesterchen, 2016-09-16:
I like the idea. I have designed a simulation for a wireless mesh network, and this one turned out to be quite helpful at our project. The simulation consisted of virtual things, virtual things gateways and virtual things access like objects, and in addition the endpoint of data, the data processing end.
This simulation has been highly modular, so we could easily disconnect the virtual endpoint and add a real one, which was great for developing new products and protocols.
The UI part - sadly - was never finished.

I am just new to TTN, so I need to figure out the basic processes first. Perhaps then we can up with a model, that does not require constant changes (at least if there are no major changes in TTN). If this simulation project then can help others to develop software/hardware without the need for actually sending any data over the net, this might be interesting for others, too.

If your idea is still lingering, send me a short message, we then could start to think of requirements.

closed 2018-10-05 by htdvisser