TheThingsArchive/ttn #90: DevAddr should be assigned by Broker, not Handler.

htdvisser, 2016-03-25:
Broker should allocate `DevAddr` and forward this to the Handler together with the join-request. Handler should use this `DevAddr` to build the join-accept.

johanstokking, 2016-04-05:
I think this should be implemented in the network controller. We should do this semantically right as we might promote the network controller to a process.

htdvisser, 2016-04-05:
Yep. I think this network controller should also be responsible for
handling MAC commands, region-specific logic etc.

htdvisser, 2016-04-06:
@johanstokking: What is our NwkAddr?

cambierr, 2016-05-25:
Hi @htdvisser !
Is there any updates on this issue ?

htdvisser, 2016-05-25:
I'm implementing this in the `refactor` branch

htdvisser, 2016-07-07:
This has been implemented in the `refactor` branch. See #199 for more info.

closed 2016-07-07 by htdvisser