TheThingsArchive/ttn #103: Assign AppEUI from the Foundation's MAC block

johanstokking, 2016-03-30:
When creating a new application, optionally assign an EUI from The Things Network Foundation's MAC block. Users should not be able to register an application with a specific AppEUI from this block.

The EUI should be generated in the Handler, not in ttnctl.

htdvisser, 2016-04-06:
@johanstokking: What is our MAC block?

johanstokking, 2016-04-06:
```
36-bit Base Value (OUI-36)	EUI-48 Address Block
70-B3-D5-7E-D	0-00 through F-FF
```

johanstokking, 2016-04-09:
We need atomic increment in a distributed way. Since we're going to use Redis anyway (see #100), we can use [INCR](http://redis.io/commands/INCR) here and convert the returned value to a byte array.

johanstokking, 2016-04-10:
We should implement this in the authentication and authorization server. The Broker trusts the token provider and thus the assigned AppEUIs. Will leave this issue here until it is closed.

PR #138 (2016-04-13): Automatic generation of AppEUI
Closes #103
closed 2016-04-13 by johanstokking