TheThingsArchive/ttn #110: Session keys after a join are incorrect

tftelkamp, 2016-03-30:
After a successful join, the nwkSkey and appSkey on the node differ from those in the backend.

htdvisser, 2016-03-30:
In the Handler they are calculated here: https://github.com/TheThingsNetwork/ttn/blob/develop/core/components/handler/handler.go#L464-L487

I don't immediately see a difference with the LoRaWAN spec. @KtorZ: do you see any mistake there?

htdvisser, 2016-03-30:
Same problem in `ttnctl join`: https://github.com/TheThingsNetwork/ttn/blob/develop/ttnctl/cmd/join.go#L145-L159

htdvisser, 2016-03-30:
Cause of this issue is probably the Endianness of the EUIs/SKeys

tftelkamp, 2016-03-31:
[LoRaWANandbyteorderingV3_923_2.xlsx](https://github.com/TheThingsNetwork/ttn/files/197128/LoRaWANandbyteorderingV3_923_2.xlsx)

htdvisser, 2016-03-31:
Fix is merged, will test this with @tftelkamp when he's available.

htdvisser, 2016-03-31:
Problem is solved.

closed 2016-03-31 by htdvisser