TheThingsArchive/ttn #117: Handling extra info in gateway status messages

jpmeijers, 2016-03-31:
The UDP packet my Multitech gateway with poly-packet-forwarder sends to TTN, contains the following JSON payload:
```
{"stat": 
  {"time":"2016-03-31.11:58:26.GMT", 
   "lati":52.23936, 
   "long":6.85647, 
   "alti":46, 
   "rxnb":0, 
   "rxok":0, 
   "rxfw":0, 
   "ackr":66.7, 
   "dwnb":0, 
   "txnb":0, 
   "pfrm":"MultiTech", 
   "mail":"my-anonymous@mail-address", 
   "desc":"Create.TTN.gateway.1"
  }
}
```
It would be nice if the fields "pfrm", "mail" and "desc" could be handled by the TTN server if they exist in the payload. Something like the e-mail address should not be publicly displayed via the API, but should be accessible via a web form in case the owner of a gateway should be contacted.

htdvisser, 2016-03-31:
Could someone comment with a link to the specific firmware that adds these fields?

gonzalocasas, 2016-03-31:
https://github.com/TheThingsNetwork/packet_forwarder/commit/faa525b1c517f481727b8b313e19a84c5bc573e7

gonzalocasas, 2016-03-31:
These fields are also supported by our [ic880a install script](https://github.com/ttn-zh/ic880a-gateway/tree/spi), so, I guess (hope) most IMST ic880a gateways do have them configured already

htdvisser, 2016-07-07:
Support for this has been implemented in the `refactor` branch. See #199 for more info.

As we will migrate to [brocaar/lora-semtech-bridge](https://github.com/brocaar/lora-semtech-bridge) to support the Semtech protocols, we still have to implement these in that project.

closed 2016-07-07 by htdvisser