TheThingsArchive/ttn #118: Add Development Mode

htdvisser, 2016-03-31:
When non-OTAA devices restart, their `FCnt` is reset. This results in messages being rejected.
When a device is configured for development mode, the backend should skip this `FCnt` check. Only question is what to do with downlink `FCnt`

johanstokking, 2016-04-09:
What should happen with the downlink frame counter is that it should reset when an uplink packet with `FCnt` 0 appears and when the device is in development mode. I don't think that we need to check for uplink frame counter roll over when the device is in development mode.

htdvisser, 2016-04-13:
Moving this issue to long-term. I think postponing this feature is a good way to encourage developers to use OTAA from the start.

closed 2016-05-06 by htdvisser