TheThingsArchive/ttn #120: Australia 915-928

tftelkamp, 2016-04-04:
Very similar to US902-928. These are differences only.

Frequency band: 915 Mhz to 928 Mhz

= Upstream
Channels 0-63
125kHz
915.2 to 927.8 in steps of 200khz

Channels 64-71
500kHz
915.9 to 927.1 in steps of 1.6Mhz

= Downstream 
Channels 0-7 
500kHz
923.3 to 927.5 insteps of 600kHz

** RX1

Downstream channel for upstream freq1 @ 125kHz:

down_chan = ((freq1-(915.2))/0.2) %% 8

Downstream channel for upstream freq1 @ 500kHz:

down_chan = ((freq1-(915.9))/1.6)

Convert to downstream frequency:

down_freq = 923.3 + (down_chan * 0.6)

DR Upstream -> DR Downstream

0 SF10BW125 - 10 SF10BW500
1 SF9BW125 - 11 SF9BW500
2 SF8BW125 - 12 SF8BW500
3 SF7BW125 - 13 SF7BW500
4 SF8BW500 - 13 SF7BW500

For reference:

8 SF12BW500 - 8 SF12BW500
9 SF11BW500 - 9 SF11BW500
10 SF10BW500 - 10 SF10BW500
11 SF9BW500 - 11 SF9BW500
12 SF8BW500 - 12 SF8BW500
13 SF7BW500 -13 SF7BW500

For now, we can distinguish Australia from the US by the uplink frequency:

902-915: US
915-928: Australia

** Sub-band

Not relevant for back-end right now. We should not use sub-band 6,7,8 because the overlap with the downlink channels.

htdvisser, 2016-04-22:
Implemented in `feature/us-frequencies`. It works, but still needs some (unit) testing before we can merge. We can field-test this on Staging.

htdvisser, 2016-04-22:
@tftelkamp: Is the TXPower the same as US915?

tftelkamp, 2016-04-26:
Just use the same TXpower as US915.

htdvisser, 2016-07-07:
Support for the AU 915MHz frequencies has been implemented in the `refactor` branch. See #199 for more info.

closed 2016-07-07 by htdvisser

dub1504, 2016-07-17:
Question:

How does end device determine RX2 channel frequency and data rate in Australian 915-928 MHz range ?