TheThingsArchive/ttn #125: Back-end sometimes fails to send ACK

johanstokking, 2016-04-05:
The device suddenly stopped receiving ACKs. It was recovered by restarting the service and the device.

The device is a mDot that was very close to the gateway. Then, quite often, exactly the packet is received twice by the same gateway.

```
router-us_1 |  DEBUG Handle PUSH_DATA                         GatewayID=008000000000B16E adapter=gateway-semtech
router-us_1 |  DEBUG Handle rxpk                              GatewayID=008000000000B16E adapter=gateway-semtech
router-us_1 |  DEBUG Handle uplink data                       GatewayID=008000000000B16E adapter=gateway-semtech
router-us_1 |  DEBUG Set duty cycles                          Frequency=912.9 GatewayID=008000000000B16E Rx1=0 Rx2=0
router-us_1 |  DEBUG Inject Metadata                          DevAddr=1D8A57AF GatewayID=008000000000B16E Metadata=Region:"us" Frequency:912.9 DataRate:"SF8BW125" CodingRate:"4/5" Timestamp:325855204 Rssi:-67 Lsnr:10 PayloadSize:15 Time:"2016-04-05T00:39:32.0474Z" RFChain:1 CRCStatus:1 Modulation:"LORA"
router-us_1 |  DEBUG Send to known brokers                    DevAddr=1D8A57AF
broker_1    |  DEBUG Handle uplink                            DevAddr=1D8A57AF
broker_1    |  DEBUG MIC check succeeded                      DevAddr=1D8A57AF DevEUI=008000000000AF02 Handler=handler:1882
broker_1    |  DEBUG Handle uplink                            DevAddr=1D8A57AF
broker_1    |  DEBUG MIC check succeeded                      DevAddr=1D8A57AF DevEUI=008000000000AF02 Handler=handler:1882
handler_1   |  DEBUG Handle Uplink                            AppEUI=0102030405060708 DevEUI=008000000000AF02
handler_1   |  DEBUG Define new bundle                        AppEUI=0102030405060708 BundleID=1-0102030405060708-008000000000AF02-00000007 DevEUI=008000000000AF02
handler_1   |  DEBUG Start buffering                          BundleID=1-0102030405060708-008000000000AF02-00000007 goroutine=set consumer
handler_1   |  DEBUG Handle Uplink                            AppEUI=0102030405060708 DevEUI=008000000000AF02
handler_1   |  DEBUG Define new bundle                        AppEUI=0102030405060708 BundleID=1-0102030405060708-008000000000AF02-00000007 DevEUI=008000000000AF02
handler_1   |  DEBUG End buffering                            BundleID=1-0102030405060708-008000000000AF02-00000007 goroutine=set consumer
handler_1   |  DEBUG Consume bundle                           BundleID=1-0102030405060708-008000000000AF02-00000007 goroutine=bundle consumer
handler_1   |  DEBUG Publishing Uplink                        AppEUI=0102030405060708 DevEUI=008000000000AF02 adapter=app-adapter
handler_1   |  DEBUG No response to send                      AppEUI=0102030405060708 DevEUI=008000000000AF02
handler_1   |  DEBUG Sending downlink packet as response      AppEUI=0102030405060708 DevEUI=008000000000AF02
broker_1    |  DEBUG Handle downlink response                 DevAddr=1D8A57AF DevEUI=008000000000AF02 Handler=handler:1882
broker_1    |  DEBUG No downlink response                     DevAddr=1D8A57AF DevEUI=008000000000AF02 Handler=handler:1882
router-us_1 |   WARN Failed forward to broker                 DevAddr=1D8A57AF error=Wrong behavior or result: Too many positive answers
router-us_1 |  DEBUG rxpk not processed                       GatewayID=008000000000B16E adapter=gateway-semtech error=Invalid operation: Too many positive answers
router-us_1 |  DEBUG Handle PULL_DATA                         GatewayID=008000000000B16E adapter=gateway-semtech
router-us_1 |  DEBUG Handle PULL_DATA                         GatewayID=0000B827EB5D8D35 adapter=gateway-semtech
```

htdvisser, 2016-04-05:
Closing this issue because I can't reproduce the problem. Please reopen if it happens again.

closed 2016-04-05 by htdvisser

johanstokking, 2016-04-12:
Fixed