TheThingsArchive/ttn #140: Downlink fails on gateways that don't send PULL_DATA

htdvisser, 2016-04-15:


avbentem, 2016-07-02:
Just FYI: it seems that some single channel gateways _do_ support downlink. But I guess you could somehow tell.

Here's a single channel NodeMCU/ESP8266 gateway that apparently supports downlink and OOTA: http://forum.thethingsnetwork.org/t/single-channel-gateway/798/174

htdvisser, 2016-07-07:
This has been implemented in the `refactor` branch. When a gateway does not indicate its downlink availability by subscribing to the downlink stream on the Router, no downlink will be scheduled on this gateway. See #199 for more info about the `refactor`.

closed 2016-07-07 by htdvisser

MoShawa, 2020-12-11:
So, what is the benift of using Single Channel Gateway ? just for monitoring purposes like: Temperature and Humidity. and can we actuate something using Arduino ?
SCG is only support one direction either: Uplink/ Downlink ?

htdvisser, 2020-12-11:
Back in 2016 the high cost of a multi-channel gateway were the biggest reason for people to build single-channel gateways. Nowadays you can get a multi-channel gateway for less than $70, so there really isn't a good reason for using single-channel gateways anymore. There is however a very good reason for not using them: **single-channel gateways are not LoRaWAN compliant**.

MoShawa89, 2020-12-11:
Thanks for your reply. Actually I am working on a Lora iot development kit comes with single channel gateway for Graduation's project. And I do not what is an architecture can be built with this system. Any suggestions? 

 Regards,
Moshawa

htdvisser, 2020-12-14:
If you have a single-channel gateway, you're technically building a LoRa network (without the WAN of LoRaWAN). The Things Network is really about LoRaWAN nowadays, so if you try to build something with a single-channel gateway, that may not work properly.

MoShawa89, 2020-12-14:
Aim to unsupported lora module on TTN. I moved to  build a server through  mqtt broker via  thingspeak cloud provider. So, i am able to send temperature & humidity to Thingspeak and meanwhile getting txt msgs from it to lora node.

Any idea for upgrading it ?

MoShawa89, 2020-12-14:
Please kindly attached this link for architecture i am not able to attach the picture from my phone.
https://www.google.com/search?q=mqtt+with+lora&client=ms-android-oppo-rvo3&prmd=vin&sxsrf=ALeKk02kc4So77w1pnbdMhFGWfwLShFVmA:1607955751053&source=lnms&tbm=isch&sa=X&ved=2ahUKEwj9iOz61c3tAhXOGewKHbdQAq0Q_AUoAnoECAcQAw&biw=360&bih=664#imgrc=cgWeSENf07p9qM