TheThingsArchive/ttn #144: Build of develop fails

kersing, 2016-04-24:
Build of develop branch fails:
> core/core.go:55: too many arguments in call to cmd.UnmarshalBinary

KtorZ, 2016-04-24:
Indeed.
Package `core` isn't currently tested, most parts are generated but still, we should add some tests about the part that are written by humans, otherwise typos like this one will sneak in again without being identified neither by Travis nor by Coveralls.

htdvisser, 2016-04-25:
Nothing wrong with `develop`. Most likely this is a problem with submodules or local dependencies.
You can run the following in a clean machine (`docker run --rm -it golang`) to verify that `develop` builds correctly:

```
git clone --recursive https://github.com/TheThingsNetwork/ttn.git $GOPATH/src/github.com/TheThingsNetwork/ttn
cd $GOPATH/src/github.com/TheThingsNetwork/ttn
make deps
make build
```

htdvisser, 2016-04-25:
I do agree with @KtorZ that we could add tests to the `core` package.

closed 2016-04-25 by htdvisser