TheThingsArchive/ttn #145: mqtt up topics do not include DEVEUI

ernstdevreede, 2016-04-25:
If I subscribe to uplink messages for an application, I get the messages from the backend, containing a lot of information, like payload, port, counter and a lot of metadata, but the EUI of the originating device is missing. This can make it difficult to interpet (decode) messages, because the keys can be coupled to the device (like in the OTAA case).
If I subscribe to  +/devices/D0509916D0D10002/up I know to which device I'm connected, but if I subscribe to +/devices/+/up the originating devices cannot be deduced from the messages.

I would propose to put one of the device identifiers in the message (probably the DEVEUI)

johanstokking, 2016-04-25:
Good point.

If you subscribe to `+/devices/+/up`, the full topic, including the `DevEUI`, is part of the topic which you can find in the metadata of the message that you receive. This requires however parsing the topic to get the `DevEUI` and that is not convenient. So I think, although redundant, it's convenient to add it.

ernstdevreede, 2016-04-25:
I wrote this issue after I tried subscribing with mosquitto_sub, which does not (normally) show topics. A python client I wrote does indeed show the topics of the messages.
So the information is there, but handling the messages would be more convenient if the information is in the messages too.

johanstokking, 2016-04-25:
Exactly. You can enable debug view with the `-d` flag in `mosquitto_sub`. But we'll add it. You're also welcome to raise a pull request.

PR #150 (2016-05-01): Add DevEUI to MQTT data as seperate item
Implement feature request "Include DevEUI in MQTT uplink payload"

Resolves #145
closed 2016-05-01 by johanstokking