TheThingsArchive/ttn #146: Frequency 869.3 not supported

htdvisser, 2016-04-26:
This is a perfectly valid frequency in the European ISM band, but it's not supported by the duty-cycle manager.

KtorZ, 2016-04-26:
Well... doesn't sound valid to me ^.^

```
    g 865.0 – 868.0 MHz 1% or LBT+AFA, 25 mW (=14dBm)
    g1 868.0 – 868.6 MHz 1% or LBT+AFA, 25 mW
    g2 868.7 – 869.2 MHz 0.1% or LBT+AFA, 25 mW
    g3 869.4 – 869.65 MHz 10% or LBT+AFA, 500 mW (=27dBm)
    g4 869.7 – 870.0 MHz 1% or LBT+AFA, 25 mW
```

htdvisser, 2016-04-26:
[image]

See http://www.etsi.org/deliver/etsi_en/300200_300299/30022001/02.04.01_40/en_30022001v020401o.pdf

tftelkamp, 2016-04-26:
865 seems a typo made by me. Apologies. It should be 863 indeed.

KtorZ, 2016-04-26:
The ETSI allow the whole range, however, the network is following the [ERC recommandations](http://www.erodocdb.dk/docs/doc98/official/pdf/rec7003e.pdf). Thus, we (I) divided the band accordingly, also advised by @tftelkamp on that point (http://thethingsnetwork.org/wiki/LoRaWAN-Frequencies)

tftelkamp, 2016-04-27:
Actually, 869.3 is an alarm band. The 'holes' in the table above are frequencies that indeed should not be used. Certainly, TTN should *never* TX at those frequencies.
MultiTech configures their gateway by default (not the TTN channel plan) to overlap with some of these alarm bands, unfortunately.

htdvisser, 2016-04-28:
Okay, thanks for this explanation. Should we allow uplink over these frequencies or keep dropping the packets, hoping that people will correctly configure their MultiTech gateways?

tftelkamp, 2016-04-29:
Let's not not accept uplinks in alarms bands, as we will not be able to do TX at those frequencies.

closed 2016-04-29 by htdvisser