TheThingsArchive/ttn #147: Response to confirmed uplink is data, not ACK.

kersing, 2016-04-26:
Using RN2483 for confirmed transmit results in 'mac_rx 1 ' response from module where it should return 'mac_tx_ok'. This indicates TTN transmits an empty data frame.

Issue listed **in bold** (tested using [gpslogger](https://github.com/kersing/go-projects/tree/master/gpslogger) and [rn2483 go library](https://github.com/kersing/go-projects/tree/master/rn2483))

Test with TTN backend:
sys reset
Expect: >RN2483<
RN2483 1.0.0 Oct 23 2015 14:46:12
mac set devaddr 020129XX
Expect: >ok<
ok
mac set appskey 2B7E151628AED2A6ABF7158809CF4F3C
Expect: >ok<
ok
mac set nwkskey 2B7E151628AED2A6ABF7158809CF4F3C
Expect: >ok<
ok
mac set adr off
Expect: >ok<
ok
mac set dr 5
Expect: >ok<
ok
mac set rx2 3 869525000
Expect: >ok<
ok
mac join abp
Expect: >ok<
ok
Expect: >accepted<
accepted
Sleep 20 secondsOpen failed: open /dev/ttyUSB2: no such file or directory
Time: , position: 0.000000, 0.000000 +- 0.000000 (,0)
mac tx cnf 1 302E3030303030302C302E303030303030
Expect: >ok<
ok
ReadResult:
**mac_rx 1**
Got line >"mac_rx 1 "<
Transmission failed: encoding/hex: odd length hex string

Test with Semtech backend (different devaddr):
sys reset
Expect: >RN2483<
RN2483 1.0.0 Oct 23 2015 14:46:12
mac set devaddr 020129XX
Expect: >ok<
ok
mac set appskey 2B7E151628AED2A6ABF7158809CF4F3C
Expect: >ok<
ok
mac set nwkskey 2B7E151628AED2A6ABF7158809CF4F3C
Expect: >ok<
ok
mac set adr off
Expect: >ok<
ok
mac set dr 5
Expect: >ok<
ok
mac set rx2 3 869525000
Expect: >ok<
ok
mac join abp
Expect: >ok<
ok
Expect: >accepted<
accepted
Sleep 20 seconds
Open failed: open /dev/ttyUSB2: no such file or directory
Time: , position: 0.000000, 0.000000 +- 0.000000 (,0)
mac tx cnf 1 302E3030303030302C302E303030303030
Expect: >ok<
ok
ReadResult:
**mac_tx_ok**
Got line >"mac_tx_ok"<
Transmission successfull

tftelkamp, 2016-04-27:
Removing the FPort field from ACKs that don't contain a downlink message should solve this?

htdvisser, 2016-05-11:
Should be easy to fix this in the `buildDownlink` func in `core/components/handler/handler.go`.

htdvisser, 2016-09-05:
Implemented correctly in `refactor` branch.

closed 2016-09-05 by htdvisser