TheThingsArchive/ttn #148: FCnt values incorrect

smartornot, 2016-04-27:
After the initialization of my node and it joined (otaa) the network...
if I use: ttnctl devices info my_DevEUI
I get for FCn:
FCntUp: 0
FCntDn: 0

That seems normal.

If my node send data (e.g. with a rn2483: "mac tx cnf 1 3939390A") I get
(after tapping again: ttnctl devices info my_DevEUI )
FCntUp: 0
FCntDn: 1

After that, if my node sends again data, I get
FCntUp: 1
FCntDn: 2

KtorZ, 2016-04-27:
The frame counter up is normally handled by your device. Are you sure that your device isn't sending '0' as a first fcnt ? Because that's exactly what it looks like.

smartornot, 2016-04-27:
I will check.
Tks for your time and patience

htdvisser, 2016-04-28:
After testing with The Things Uno, I can confirm that the first frame counter sent by the device is indeed `0`

closed 2016-04-28 by johanstokking