TheThingsArchive/ttn #157: Support for encrypted MQTT messaging

tedvanderveen, 2016-05-11:
Implement MQTT protocol with support for TLS/SSL encryption over port 8883.

krital, 2016-05-11:
Ok you can assign it to me if you want

krital, 2016-05-22:
Hi all, i have just found time to do this. You can get it on https://github.com/krital/mosquitto_tls / https://hub.docker.com/r/krital/mosquitto_tls/

johanstokking, 2016-05-23:
Thanks. We've got it working already, this issue should have been closed.

It's a nice implementation but we need externally generated and managed keys in our scenario. Also we're moving to a custom, slimmer base image and build.

closed 2016-05-23 by johanstokking

krital, 2016-05-23:
wish i knew so that i didn't waste time trying to help. I might go back into the letsencypt version now.

johanstokking, 2016-05-23:
It's still valuable @krital for the community. The security of Mosquitto really depends on the use case. I think this is very usable in private networks that have expose the MQTT broker.