TheThingsArchive/ttn #159: Application events on MQTT

htdvisser, 2016-05-11:
Applications should be able to publish more events to MQTT in addition to activations.

- [ ] Downlink expired
- [ ] Downlink sent

htdvisser, 2016-09-22:
Updated list of events:

* Downlink Sent (https://github.com/TheThingsNetwork/node-app-lib/issues/11)
* Downlink ACKed (https://github.com/TheThingsNetwork/ttn/issues/158)
* Payload Function Error (https://github.com/TheThingsNetwork/ttn/issues/254)

All of these are device events, so I suggest to publish these to the `<app-id>/devices/<dev-id>/events` topic. For application events we could add the `<app-id>/events` topic. 

Comments, @johanstokking?

johanstokking, 2016-09-22:
No comments. Go go go

htdvisser, 2016-09-22:
Done

commit 37f8ebb3 (2016-09-22): Publish device events to MQTT

- Failed Activations
- Failed Uplinks
- Failed Downlinks
- Scheduled Downlinks
- Sent Downlinks
- ACKed Downlinks (note that downlinks are still sent as unconfirmed)

Closes #159
closed 2016-09-22 by htdvisser