TheThingsArchive/ttn #165: Something wrong with ttnctl

DIY3D, 2016-05-19:
Hi,

I try to create a user as I did before but I got this error:

C:\ttnctl-windows-amd64>ttnctl user create <my emailaddress>
Password: <my password>
Confirm password: <my password>
←[31m FATAL←[0m Registration failed: 500 500 Internal Server Error

regards,
Marco

htdvisser, 2016-05-19:
Thanks for submitting the issue. I looked into the problem you're experiencing.

First of all, I would encourage you to download a newer version of `ttnctl`. It looks like yours is a little old (we removed the `←[31m` and `←[0m ` from the Windows version a few weeks ago).

The error you see is caused by a duplicate entry in the database (you already registered with this email address). Apparently the upstream server is not very good in explaining what's wrong, so I will file an issue there to make the error messages more descriptive.

P.S. I'm pretty sure `iot_marco@thethingsnetwork.org` is not your real email address.

DIY3D, 2016-05-19:
Beste Hylke,
Dank voor je reactie, heb ttnctl inmiddels geupdate en het werkt nu (met het juiste emailadres ;-)
 Misschien nog een aandachtspunt, ik kreeg een melding van F-secure die ik bij de vorige versie van ttnctl niet kreeg (of me ontgaan is).

              

groet,Marco

Date: Thu, 19 May 2016 00:26:00 -0700

johanstokking, 2016-05-20:
Hi Marco, please write in English and stay on-topic.

Can you file a separate issue of the _specific_ F-secure alert that you got?

htdvisser, 2016-05-25:
See #169

closed 2016-05-25 by htdvisser