TheThingsArchive/ttn #168: 403 Forbidden when managing applications

htdvisser, 2016-05-21:
```
❯ ttnctl user login demo@thethingsnetwork.org
Password:
  INFO Logged in as demo@thethingsnetwork.org and persisted token in /Users/htdvisser/.ttnctl/auths.json

❯ ttnctl applications
  INFO Found 1 application(s)
EUI             	Name           	Owner                    	
70B3D57ED0000121	Hello World App	demo@thethingsnetwork.org

❯ ttnctl applications authorize 70B3D57ED0000121 hylke@thethingsnetwork.org
 FATAL Failed to authorize user: 403 Forbidden
```

romeovs, 2016-05-21:
This is caused by `ttnctl` not serializing the app EUI correctly. When I look at the logs in the auth server for:

```
ttnctl applications authorise 000000000000006 romeo@thethingsindustries.com
```

I see a call to:
```
/applications/38303030303030303030303030303036/authorize
```
This is usually caused by doubly converting the app EUI to hex (so this is the hex of the the encoded version of the app EUI).

romeovs, 2016-05-21:
I was making a PR for this but accidentally pushed it on `develop`. Check the commit [here](https://github.com/TheThingsNetwork/ttn/commit/294996b60230891069c2c3d401065b6c43e24bbe).

htdvisser, 2016-05-21:
Thanks, @romeovs! Downloading new `ttnctl` resolves this issue.

closed 2016-05-21 by htdvisser