TheThingsArchive/ttn #170: Can't login after logout

DIY3D, 2016-05-24:
C:\ttnctl-windows-amd64>ttnctl user
  WARN No login found. Please login with ttnctl user login [e-mail]

C:\ttnctl-windows-amd64>ttnctl user login [my email] 
Password:[my password]
 FATAL Failed to login                          error=User not found

htdvisser, 2016-05-25:
Thank you for submitting this issue. I'll work with you to figure out what's going wrong here.

Could you please let me know which version of `ttnctl` you are using by submitting the output of `ttnctl version`.

johanstokking, 2016-07-06:
@DIY3D we haven't heard from you for a while. If you still encounter this issue, please raise a new one, mentioning the output of `ttnctl version`. Also note that your account should be valid by having your e-mail address confirmed. Also try the dashboard; https://staging.thethingsnetwork.org

closed 2016-07-06 by johanstokking