TheThingsArchive/ttn #172: Issues with TTN dashboard

cambierr, 2016-05-30:
Since there is no repo yet for the online dashboard, here are some issues I spotted:

- [ ] Deleting an application containing "__" (two consecutive underscores) always fails with a 500
- [ ] The JWT refresh seems to always fail with a 401

romeovs, 2016-05-30:
Thanks for the report!
The second issues has already been identified and is being worked on.
I'll also be looking into the first one.

cambierr, 2016-05-30:
Cool... I'll keep adding bugs & suggestions on my first mesage to avoid creating a lot of issues in this repo :)

romeovs, 2016-05-30:
thanks! Some more notes on your issues:
- Documentation is forthcoming
- The auth server will get a big rewrite soon

cambierr, 2016-05-31:
If you are looking for contributors to the new auth backend, I'm totally in!

cambierr, 2016-06-02:
Isn't the code of the UI or auth/api backends available on github?

johanstokking, 2016-06-02:
@cambierr it's not open source. The future plans of the dashboard are specific functionality for the public community network: e.g. optional quality of service and value compensation, possibly using a cryptocurrency, hosted third-party integrations, a hosted device application compiler, etc. For private networks, just use `ttnctl` or BYOD (bring your own dashboard)

The dashboard just calls the public endpoints just like the (open source) `ttnctl` command line utility does. The account server is an OAuth 2.0 token issuer. Nothing special really

cambierr, 2016-06-02:
Okay cool. We are already using our own dashboard using the public api for now (since external handlers are not yet supported) but I was looking for a list of the existing endpoints (our current prototype has been done using network analyzer of browsers to find endpoints which is not the best way).

Anyway, thank you for your feedback.

johanstokking, 2016-06-02:
I see. Those are in this repository in fact, not in the dashboard. We're refactoring this to get the separation of concerns right, so they will change and that's why they're not documented. Keep an eye on this: https://github.com/TheThingsNetwork/ttn/tree/refactor/api

cambierr, 2016-06-03:
I added a few issues ;-)

romeovs, 2016-06-03:
@cambierr Thanks! I've numbered them to make them more discussable. Some notes:

**1.** I've been unable to reproduce this so far. Can you see wether or not there is an HTTP error, and if so, what that error is?
**2.** This should be fixed with a recent update to the account server. If you're still having this problem, please notify me. The problem was that the account server removed all refresh tokens of a user for that client when issuing new token pairs, so different browsers would interfere with each other.
**3.** We should switch to implicit grant flow on `ttnctl`, this is not an issue with the dashboard however. Please make a different issue for this.
**7.** The account server only redirects you to the dashboard if you were already coming from the dashboard. This makes it possible to have more than one client use the account server for issuing tokens (like the Wiki/community pages for example) in the future. We will be building a hub of some kind to show to the user after he just logs in without trying to log in to any specific service.

cambierr, 2016-06-03:
1. I can't reproduce it anymore. I guess it has been fixed in one of your updates ?
2. I will do some testing and be back to you
3. Ok, I'll create an issue about that
7. Has been fixed in fact (when I added it there was no redirection url. I update my first post)

Also, what about adding a totp security to the auth backend ? :)

johanstokking, 2016-07-06:
Hi @cambierr 

What's the status on 2, 3 and 6 on your side?

Regarding 4 and 8: We're merging a new model for applications soon, as part of the refactored back-end. We'll add documentation about interfaces to address 4, and there will be a nicer response than 500 when there are no payload functions set to address 8.

The new model has an unique App ID field as primary key, while the AppEUI will be free to define. Users may use their own AppEUIs, share them or issue a new one. The authorization will be based on App ID, instead of AppEUI. We're migrating existing applications to use the AppEUI as AppID.

cambierr, 2016-07-06:
@honorem is the guy who creates the temp connector between our backend and the current ttn architecture (no external handlers) so I let him give some feedback.

romeovs, 2016-07-06:
Let me chime in as well:

- *2.* This should not be happening anymore
- *6.* The dashboard schedules its own token refreshes, and refreshes the token on visit, so this should no longer be happening (please check this).

closed 2016-08-16 by cambierr

paultanner, 2017-01-06:
6: "You do not have any applications" still happening.  In the case in point we have not used the service for at least a month so it's likely that tokens have expired.

romeovs, 2017-01-06:
@paultanner If I understand correctly you had applications before and now they're gone right?
Can you have a look in your browsers developer console for errors?

paultanner, 2017-01-06:
thx @romeovs, yes my app has disappeared.  console log looks fairly ok

```
[Log] dispaching socket.connect – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after socket.connect – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Debug] creating socket (main.63af8b88c96ddae1789a.js, line 2)
[Log] dispaching token.update – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after token.update – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Debug] socket accepting new token (main.63af8b88c96ddae1789a.js, line 2)
[Log] dispaching token.refresh – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after token.refresh – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Debug] scheduling token refresh in 247049ms (main.63af8b88c96ddae1789a.js, line 7)
[Log] dispaching @@reduxReactRouter/routerDidChange – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after @@reduxReactRouter/routerDidChange – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching @@reduxReactRouter/initRoutes – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after @@reduxReactRouter/initRoutes – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching handlers.list – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after handlers.list – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching applications.list – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after applications.list – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Debug] emitting – "nowhere" (main.63af8b88c96ddae1789a.js, line 2)
[Log] React rendered! (main.63af8b88c96ddae1789a.js, line 4)
[Log] dispaching applications.list.success – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after applications.list.success – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching handlers.upsert – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after handlers.upsert – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching handlers.upsert – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after handlers.upsert – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching handlers.upsert – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after handlers.upsert – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching handlers.upsert – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after handlers.upsert – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching handlers.list.success – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching applications.list – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after applications.list – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after handlers.list.success – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Debug] socket connected (main.63af8b88c96ddae1789a.js, line 2)
[Log] dispaching socket.connected – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after socket.connected – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] dispaching applications.list.success – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
[Log] state after applications.list.success – Object (main.63af8b88c96ddae1789a.js, line 1)
Object
```
.. apart from the couple of lines in the middle. they may give you a clue.