TheThingsArchive/ttn #175: Dashboard: clicking on ABP devices button doesn't work

gonzalocasas, 2016-06-03:
This was working earlier today (I think), but it looks like a recent change broke it.

Repro:
1. Open dashboard
2. Click on an app
3. Click on the "ABP" button on the "Devices" section
 -> error reports on console

[image]

romeovs, 2016-06-03:
Thanks for the report. Can you tell me what browser version you are using?
Does this happen every time? Can you post the full stack trace of the error?

gonzalocasas, 2016-06-03:
Chrome 50 on Mac OS, and also Firefox.
It happens every time. (but it was working earlier, so I assume a recent deployment broke it)

Here's the stack trace:
```
module.exports (main.js:27427)
module.exports (main.js:28063)
from (main.js:28083)
exports.default (main.js:33616)
exports.default (main.js:33563)
ReactCompositeComponentMixin._constructComponentWithoutOwner (main.js:14030)
ReactCompositeComponentMixin._constructComponent (main.js:14016)
ReactCompositeComponentMixin.mountComponent (main.js:13939)
ReactCompositeComponent_mountComponent (main.js:7230)
ReactReconciler.mountComponent (main.js:7305)
ReactMultiChild.Mixin.mountChildren (main.js:13270)
ReactDOMComponent.Mixin._createInitialChildren (main.js:10263)
ReactDOMComponent.Mixin.mountComponent (main.js:10109)
ReactDOMComponent_mountComponent (main.js:7230)
ReactReconciler.mountComponent (main.js:7305)
ReactCompositeComponentMixin.performInitialMount (main.js:14077)
ReactCompositeComponentMixin.mountComponent (main.js:14002)
ReactCompositeComponent_mountComponent (main.js:7230)
ReactReconciler.mountComponent (main.js:7305)
ReactMultiChild.Mixin.mountChildren (main.js:13270)
ReactDOMComponent.Mixin._createInitialChildren (main.js:10263)
ReactDOMComponent.Mixin.mountComponent (main.js:10109)
ReactDOMComponent_mountComponent (main.js:7230)
ReactReconciler.mountComponent (main.js:7305)
ReactMultiChild.Mixin.mountChildren (main.js:13270)
ReactDOMComponent.Mixin._createInitialChildren (main.js:10263)
ReactDOMComponent.Mixin.mountComponent (main.js:10109)
ReactDOMComponent_mountComponent (main.js:7230)
ReactReconciler.mountComponent (main.js:7305)
ReactCompositeComponentMixin.performInitialMount (main.js:14077)
ReactCompositeComponentMixin.mountComponent (main.js:14002)
ReactCompositeComponent_mountComponent (main.js:7230)
ReactReconciler.mountComponent (main.js:7305)
ReactMultiChild.Mixin.mountChildren (main.js:13270)
ReactDOMComponent.Mixin._createInitialChildren (main.js:10263)
ReactDOMComponent.Mixin.mountComponent (main.js:10109)
ReactDOMComponent_mountComponent (main.js:7230)
ReactReconciler.mountComponent (main.js:7305)
ReactMultiChild.Mixin.mountChildren (main.js:13270)
ReactDOMComponent.Mixin._createInitialChildren (main.js:10263)
ReactDOMComponent.Mixin.mountComponent (main.js:10109)
ReactDOMComponent_mountComponent (main.js:7230)
ReactReconciler.mountComponent (main.js:7305)
ReactCompositeComponentMixin.performInitialMount (main.js:14077)
ReactCompositeComponentMixin.mountComponent (main.js:14002)
ReactCompositeComponent_mountComponent (main.js:7230)
ReactReconciler.mountComponent (main.js:7305)
ReactMultiChild.Mixin._mountChildAtIndex (main.js:13446)
ReactMultiChild.Mixin._updateChildren (main.js:13366)
ReactMultiChild.Mixin.updateChildren (main.js:13326)
ReactDOMComponent.Mixin._updateDOMChildren (main.js:10480)
ReactDOMComponent.Mixin.updateComponent (main.js:10326)
ReactDOMComponent.Mixin.receiveComponent (main.js:10282)
ReactDOMComponent_receiveComponent (main.js:7230)
ReactReconciler.receiveComponent (main.js:7369)
ReactChildReconciler.updateChildren (main.js:13617)
ReactMultiChild.Mixin._reconcilerUpdateChildren (main.js:13245)
ReactMultiChild.Mixin._updateChildren (main.js:13338)
ReactMultiChild.Mixin.updateChildren (main.js:13326)
ReactDOMComponent.Mixin._updateDOMChildren (main.js:10480)
ReactDOMComponent.Mixin.updateComponent (main.js:10326)
ReactDOMComponent.Mixin.receiveComponent (main.js:10282)
ReactDOMComponent_receiveComponent (main.js:7230)
ReactReconciler.receiveComponent (main.js:7369)
ReactChildReconciler.updateChildren (main.js:13617)
ReactMultiChild.Mixin._reconcilerUpdateChildren (main.js:13245)
ReactMultiChild.Mixin._updateChildren (main.js:13338)
ReactMultiChild.Mixin.updateChildren (main.js:13326)
ReactDOMComponent.Mixin._updateDOMChildren (main.js:10480)
ReactDOMComponent.Mixin.updateComponent (main.js:10326)
ReactDOMComponent.Mixin.receiveComponent (main.js:10282)
ReactDOMComponent_receiveComponent (main.js:7230)
ReactReconciler.receiveComponent (main.js:7369)
ReactCompositeComponentMixin._updateRenderedComponent (main.js:14441)
ReactCompositeComponentMixin._performComponentUpdate (main.js:14423)
ReactCompositeComponentMixin.updateComponent (main.js:14352)
ReactCompositeComponent_updateComponent (main.js:7230)
ReactCompositeComponentMixin.receiveComponent (main.js:14275)
ReactReconciler.receiveComponent (main.js:7369)
ReactCompositeComponentMixin._updateRenderedComponent (main.js:14441)
ReactCompositeComponentMixin._performComponentUpdate (main.js:14423)
ReactCompositeComponentMixin.updateComponent (main.js:14352)
ReactCompositeComponent_updateComponent (main.js:7230)
ReactCompositeComponentMixin.performUpdateIfNecessary (main.js:14291)
ReactReconciler.performUpdateIfNecessary (main.js:7388)
runBatchedUpdates (main.js:6923)
Mixin.perform (main.js:7719)
Mixin.perform (main.js:7719)
_assign.perform (main.js:6869)
flushBatchedUpdates (main.js:6945)
ReactUpdates_flushBatchedUpdates (main.js:7230)
Mixin.closeAll (main.js:7785)
Mixin.perform (main.js:7732)
ReactDefaultBatchingStrategy.batchedUpdates (main.js:16024)
batchedUpdates (main.js:6877)
ReactEventListener.dispatchEvent (main.js:16184)
```

gonzalocasas, 2016-06-07:
Today I realized something: it only happens with one of my two applications!
Here I can reproduce every time: https://staging.thethingsnetwork.org/applications/70B3D57ED00001D3
but with another Application, it works just fine.

romeovs, 2016-06-07:
Okay. We just pushed a new version of the dashboard that has a lot of small fixes in it, including more sanity checks.

Can you check if it still occurs?

gonzalocasas, 2016-06-07:
I just checked. Still occurs.

romeovs, 2016-07-06:
Found the error! You have a device which has a device address `00000000`. The JSON serialisation on the server considers this as an empty address and doesn't send it when listing devices.
In the dashboard, the address was expected however, so the UI crashed.

I've fixed the dashboard to return a device EUI based on the device address for ABP devices, and send it even when the EUI 'empty'.

 Can you verify this works now?

gonzalocasas, 2016-07-06:
Works! Thanks, you're awesome!

romeovs, 2016-07-06:
Nice!

closed 2016-07-06 by romeovs