TheThingsArchive/ttn #183: MQTT Data overhead

trlafleur, 2016-06-09:
I might suggest that most sensor nodes only need a minimal amount of data delivered to an application. The current MQTT payload package overhead is large with over 350 bytes of data. 

I think we should break this down to 4 classes as defined when an application is set up, or broken down in the MQTT delivery message as an option that one could just subscribe too.

1) FP,  full package as we have now.
2) MP,  medium package, with just the: {"payload":"SGVsbG8sIFdvcmxkIQ==","counter":1699,","gateway_timestamp":3220137616} or other data??
3) SP, a minimal package with just the payload:  {"payload":"SGVsbG8sIFdvcmxkIQ=="}
4) DATA, a minimal package with just the payload un-encrypted: {"payload":"abcdefghijklmnop"}

This would allow small device to subscribe to MQTT to get data without the overhead. It would allow sensor node to subscribe to get message from other nodes.

thanks

johanstokking, 2016-06-09:
Thanks, good suggestion.

We've been brainstorming about posting the `fields`, the result of the payload functions (not really well documented yet, see [dashboard](https://staging.thethingsnetwork.org/dashboard)), as topics.

For example, if your payload is `09 AC 00 51`, and your payload functions decode it as:
```json
{
  "temperature": 24.7,
  "luminosity": 81
}
```

Then you might be interested in the following publishes:

- `<AppEUI>/devices/<DevEUI>/up/raw` with `09 AC 00 51` (as bytes)
- `<AppEUI>/devices/<DevEUI>/up/temperature` with `24.7`
- `<AppEUI>/devices/<DevEUI>/up/luminosity` with `81`
- `<AppEUI>/devices/<DevEUI>/up` with everything, just like now: `{ "payload": "CawAUQ==", "fields": { "temperature": 24.7, "luminosity": 81 }, "counter": 123, ... }`

What do you think?

BTW 1, actually what I think is ideal, is that MQTT had support for arbitrary metadata, and that on subscribe, a client can provide metadata fields that it is interested in. The message is then just the payload, and the metadata is selected per subscriber. Now, we'd have to split it ourselves, and you're suggesting that we pick relevant fields for clients, but I'm afraid that it ends in a subjective discussion cause it's not one size fits all.

BTW 2, regarding 4), the payload is not encrypted, it is just base64 encoded. We can't convert that to plain text because, hopefully, devices don't send plain text. However, MQTT messages are in fact bytes, so we could post the raw payload as suggested above, as is.

trlafleur, 2016-06-09:
My message was just to start people thinking on how to deliver the payload...

I do like your idea of being able to just subscribe to arbitrary metadata... good option!

Yes, I was un-aware that it was just base64 encoded, I need to read more documentation! 
 
Sending just bytes is a good compromise... easy to parse in a small device.

closed 2016-07-06 by johanstokking

4flixt, 2019-11-24:
Is this feature still working as desired? I was subscribing to something like
```<AppEUI>/devices/<DevEUI>/up/temperature```
which worked until a couple of days back. I checked and the full messages are still received via MQTT if I sub to:
```<AppEUI>/devices/<DevEUI>/up```
which is the case. MQTT explorer also wasnt able to directly obtain the different payload fields anymore (this used to work, when I set everything up maybe 4 weeks ago).

johanstokking, 2019-11-25:
We didn't change anything there. What happens if you subscribe to `<AppEUI>/devices/<DevEUI>/up/#`?

nottheedge, 2019-11-25:
Could it be related to that -> [TTN in maintenance ](https://status.thethings.network/)

4flixt, 2019-11-25:
Yes! Thank you. I also just found this today.