TheThingsArchive/ttn #188: TTN no login possible, dshboard and ArduinoUno, any issues?

EvertSchut, 2016-06-27:
Hi, I'm new to TTN and it seems all troubles for me now. My TTNAruinoUno was succesfully activated in the meetup in Utrecht at june 14, 2016 but joining now does not work.

(1) Trying to connect to TTN did not with my TTN Arduino Uno work and now (2) even log in to dashboard does not work anymore.

1: The TTN Dasboard is unreachable for me:
- The site in https://www.thethingsnetwork.org/labs/story/the-things-uno-getting-started/step/connect-with-otaa "2. Register device / 1. Go to The Things Network Dashboard. 2. Create an account and/or log-in. results in "502 Bad Gateway".
- I try to login at https://www.thethingsnetwork.org/login/ it seems mij usename and password dont match
- But also the https://www.thethingsnetwork.org/accounts/password/reset/ does not result in an email.
Are there issues?

2: Also getting to **connect tot TTN does not work here ** -  for days now.  Even modified some code now I see that the result is:

EUI: 0004A30B001B4A7E
Battery: 3273
AppEUI: 70B3 ..(andsomeore)
DevEUI: 0004A30B001B4A7E
DevAddr: 00000000 (isn't this strange??)
Band: 868
Data Rate: 5
RX Delay 1: 1000
RX Delay 2: 2000
Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B4A7E
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Join not accepted
denied 

The "denied" is the result of some coding, coming from debugPrintLn(response) and making ttu in sthe sketch available. I seem to het an RF path and answer, but not a connection / acceptance.
Is this also some network issue? 

(I noticed the 00000000 DevAdrr, wich seems strange to me but maybe it is like DHCP assigned after a succesful join?).

Thanks for any help, Evert

romeovs, 2016-06-27:
@EvertSchut, thanks for the bug report. It is pretty big an touches a couple of topics so let me try and help you out:

**1.1**
I've tried reproducing errors you get on the dashboard but everything seems to be working here. (both logging in and registration work for me). Could you please provide more details on exactly *when* you get the 502 error?

**1.2** and **1.3**
We're working on a global Single-Sign On for all our services. This is not in production however so creating an account at https://account.thethingsnetwork.org unfortunately does not grant you access to https://www.thethingsnetwork.org. As I've said we're working on moving all of this to one account, but for now you'll have to create multiple accounts. When the time comes to move to one account we will merge your accounts for you. For the same reason the password reset will not work.

Can you try logging in with the new account at https://account.thethingsnetwork.org?

2. This is hard to debug without looking at your code, do you have a minimal working example that reproduces this error?

EvertSchut, 2016-06-28:
Hi Romeo, thanks for your fast response. 

Just tried, now a new day, another location "https://account.thethingsnetwork.org/users/authorize?client_id=ttn-dashboard&redirect_uri=https%3A%2F%2Fstaging.thethingsnetwork.org%2F&response_type=code&state=tkyohigYVG%2B%2Fapplications" 
.. but again 502 bad gateway message.

The code: 

First the Modified OTAA example (Sketch in Arduino cc):

#include "TheThingsUno.h"

// Set your AppEUI and AppKey
const byte appEui[8] = {  0x70, 0xB3, 0xD5, 0x7E, 0xD0, 0x00, 0x04, 0x05  }; //for example: {0x70, 0xB3, 0xD5, 0x7E, 0xE0, 0xE0, 0x01, 0x4A1};
const byte appKey[16] = { 0x87, 0xE4, 0xA4, 0x2D, 0xD7, 0x9A, 0x12, 0xCC, 0x05, 0xD4, 0x66, 0x9D, 0xE1, 0x9A, 0x43, 0xC0 };
#define debugSerial Serial
#define loraSerial Serial1

// Set your message to send
String message = "Hello world"; //sending a string of chars "Hello world"

#define debugPrintLn(...) { if (debugSerial) debugSerial.println(__VA_ARGS__); }
#define debugPrint(...) { if (debugSerial) debugSerial.print(__VA_ARGS__); }

TheThingsUno ttu;
// vereist eigen .h file op C waarbij definities in 6 regels (readvalue, sendCommand, readLine etc) 
// van private naar public zijn gemaakt!
bool eigenjoin(const byte appEui[8], const byte appKey[16]) {
    ttu.showStatus();
String devEui;
String deBuginfo = "";
devEui = ttu.readValue("sys get hweui");
  ttu.sendCommand("mac set appeui", appEui, 8); 
  // werkt niet omdat const byte is geen String: deBuginfo = deBuginfo + String(appEui);
  ttu.sendCommand("mac set deveui " + devEui);
  ttu.sendCommand("mac set appkey", appKey, 16);
  ttu.sendCommand("mac join otaa");

  String response = ttu.readLine(10000);
  if (response != "accepted") {
    debugPrintLn("Join not accepted");
    debugPrintLn(response);  // lees de status hack 
    // wacht even
    // tegen het floodden van het netwerk
    int i=0;
    while ( i<10)  {  
      debugPrint(".");
      delay(2000); // tegen het floodden van het netwerk 
      i=i+1;  
      }
      
    return false;
  }

  debugPrintLn("Join accepted: " + response + ". Status: " + ttu.readValue("mac get status"));

Serial.print("eigen joint \n");
return true;
}

// hier begint de Main

void setup() {
  debugSerial.begin(115200);
  loraSerial.begin(57600);

  delay(2000);
Serial.print("Hier begint ie! \n");

  ttu.init(loraSerial, debugSerial);
  ttu.reset();

  //the device will attempt a join every second till the join is successfull
  while(!eigenjoin(appEui, appKey)){
    Serial.print("next joining attempt..........\n");
    ttu.showStatus();
      delay(2000);
  }

  digitalWrite(13, HIGH); //turn on LED to confirm join

// debugPrintLn("debug wordt uitgevoerd");

  delay(6000);
  ttu.showStatus();
  debugPrintLn("Setup for The Things Network complete");

  delay(1000);
}

void loop() {

  ttu.sendString(message);
  delay(1000);
}

- - - - - - - - - - 

Then to be able to use some commands, we (my colleague and I) modified TheThingsUno.h:

class TheThingsUno
{
  private:
    Stream* modemStream;
    Stream* debugStream;
    String model;

//    String readLine(int waitTime = DEFAULT_WAIT_TIME);
//    bool waitForOK(int waitTime = DEFAULT_WAIT_TIME, String okMessage = "ok");
//    String readValue(String key);
//    bool sendCommand(String cmd, int waitTime = DEFAULT_WAIT_TIME);
//    bool sendCommand(String cmd, String value, int waitTime = DEFAULT_WAIT_TIME);
//    bool sendCommand(String cmd, const byte* buf, int length, int waitTime = DEFAULT_WAIT_TIME);
//    bool enableFsbChannels(int fsb);

  public:
    int downlinkPort;
    byte downlink[64];
    void init(Stream& modemStream, Stream& debugStream);
    void reset(bool adr = true, int sf = DEFAULT_SF, int fsb = DEFAULT_FSB);
    bool personalize(const byte devAddr[4], const byte nwkSKey[16], const byte appSKey[16]);
    bool join(const byte appEui[8], const byte appKey[16]);
    int sendBytes(const byte* buffer, int length, int port = 1, bool confirm = false);
    int sendString(String message, int port = 1, bool confirm = false);
    void showStatus();
// begin mods 27-6-2016
	String readValue(String key);
	bool sendCommand(String cmd, int waitTime = DEFAULT_WAIT_TIME);
	bool sendCommand(String cmd, String value, int waitTime = DEFAULT_WAIT_TIME);
	bool sendCommand(String cmd, const byte* buf, int length, int waitTime = DEFAULT_WAIT_TIME);
	bool enableFsbChannels(int fsb);
	String readLine(int waitTime = DEFAULT_WAIT_TIME);
	 bool waitForOK(int waitTime = DEFAULT_WAIT_TIME, String okMessage = "ok");
	
};

#endif

EvertSchut, 2016-06-28:
@romeovs See previous post, forgot to mention you and also closed it, jus learning here. I just tried another browser, got error with IE 400 https://account.thethingsnetwork.org/users/authorize. Previous attempts were on FF. I Shouldn' be the only one with this problem, though? I'll try another machine.

romeovs, 2016-06-28:
@EvertSchut you're getting the hang of it!
I've located the problem that was causing the 502 errors, and fixed it. Can you try again?The correct url is https://account.thethingsnetwork.org/users/login

EvertSchut, 2016-06-28:
That results via FF in De pagina verwijst niet op een juiste manier door

Firefox heeft vastgesteld dat de server de aanvraag voor dit adres doorverwijst op een manier die nooit zal eindigen.

    Dit probleem kan soms worden veroorzaakt door het uitschakelen of weigeren van cookies.

EvertSchut, 2016-06-28:
zal ff proberen met coookies

EvertSchut, 2016-06-28:
Cookies lijken aan te staan en ABP uitgezet maakt ook niet uit.

EvertSchut, 2016-06-28:
Using IE I got a message that I was logged in, got redirected to https://staging.thethingsnetwork.org/ and that is an empty screen, a non reacting page also when I reload it.

romeovs, 2016-06-28:
Can you tell me what OS and version of IE / Firefox you are using?

EvertSchut, 2016-06-28:
Browsers here on my laptop: FF 47.0 or IE 11.0.9600. But I just tried a diffrent machine, it seems that this machine gains access. So I'm afraid that something is wrong on my machine / laptop here. I will close all tabs (got 40 open or so) and I'm running an extra virus scan now. 
Strange thing is that the I userd the laptop in the meeup session and all was working fine until I logged out from the dashboard, after that I could not get in anymore. I'll look into the laptop now.

The other problem was that my TTNAruinuinoOne could not connect. Any hintson 
Sending: mac join otaa
Join not accepted
denied ?

romeovs, 2016-06-28:
Ok - I do not think you need to worry to much about viruses! We had an issue recently where a truncated version of the scripts we use on the dashboard were sent, causing this white page you're encountering. Maybe this wrong script is still in your browser cache?

I'm not a pro on why the join might not work so I added a colleague of mine to the discussion, he might be able to tell you more.

EvertSchut, 2016-06-28:
Browser / laptop problem solved, freed up 9 GB on C:, partly using Crap Cleaner, cleared caches, renewed Firefox... got access to the Dashboard and could log in with the link you've sent.

Any hints on the TTNArduinoUno wich connected june 14 in a meetup, and now doesn't join? The sketch from the meetup was not saved and is the above sketch. Checked the keys 5 times, maybe I make a mistake every time.

Output:
Sending: mac join otaa
Join not accepted
denied

Any hints?

johanstokking, 2016-06-28:
Are you close to a gateway? Which one?

EvertSchut, 2016-06-28:
Driebergen, I presume.

I tried a page (from semtech i thougt) but did not see any reception, I
didn see any reception at all.

Also there I got some problems creating an account since gmail was not
allowed and anther domain was allowed, but I did not receive the acknoledge
mail on that account.

johanstokking, 2016-06-28:
OK, that's not TTN but Semtech.

The Things Uno's reach, without external antenna, is very limited. You need to be close to a gateway for it to work.

I'm closing this issue as there's at this point not much we can do. Please file an issue with a specific problem you're facing, and if there are more topics, create additional issues.

closed 2016-06-28 by johanstokking