TheThingsArchive/ttn #189: Docker compose file problems

eddiec, 2016-06-28:
Hi, using the compose file in http://staging.thethingsnetwork.org/wiki/Backend/Docker
gives me the error:

$ docker-compose up
ERROR: The Compose file './docker-compose.yml' is invalid because:
services.redis.ports contains an invalid type, it should be an array

I'm on "CentOS Linux release 7.2.1511 (Core)" with "docker-compose version 1.7.1, build 0a9ab35" (the latest version for centos AFAICT).

Uncommenting the line obviously allowed progress but much confusion ares from the entry further down:

johanstokking, 2016-06-28:
Comment `ports:` helps I think

johanstokking, 2016-06-28:
BTW, you can also use `- "127.0.0.1:6379:6379" to expose the port to the host machine. This allows you to use `redis-cli`, while not exposing the port publicly.

The `TTN_HANDLER_REDIS_ADDR=redis:6379` refers to the `redis` service. It resolves to the IP address of that Docker container. Exposing the port doesn't matter to make it accessible for other Docker containers when using Docker compose.

eddiec, 2016-06-28:
> The  TTN_HANDLER_REDIS_ADDR=redis:6379  refers to the  redis  service. It resolves to the IP address of that Docker container.

OK but when I run the compose command I get an error because it tries to connect to redis.my.domain which happens to exist but is a different host.  So a conflict somewhere maybe?

This issue is mostly an FYI (I should probably learn about labels!) I should add.

johanstokking, 2016-06-28:
OK, this is more a general FYI issue regarding Docker.

Then again, you can point it to your host, without exposing the port publicly, see the first part of my second comment.

closed 2016-06-28 by johanstokking