TheThingsArchive/ttn #194: Decorate device activation event

johanstokking, 2016-07-06:
When device activation using LoRaWAN, add `DevAddr`, `AppSKey` and `NwkSKey` to update monitors live

romeovs, 2016-07-06:
Also add the `DevEUI`, this is nice to know when listening to activations for all devices of an application.

johanstokking, 2016-07-06:
The `DevEUI` is part of the topic. Requires some parsing but the topic structure is stable.

htdvisser, 2016-07-06:
> _topic structure is stable_

This is not entirely accurate, as the topic format will change with the new application structure. Is there already a definition for this, @johanstokking?

johanstokking, 2016-07-06:
Stable as in it remains `<app identifier>/devices/<device identifier>/[up|down|activations]`. We are indeed changing the app identifier from `AppEUI` to a self-defined but immutable identifier.

If we do that also for devices, then `DevEUI` really becomes LoRaWAN specific and it should be added to the metadata, next to `DevAddr`, `NwkSKey` and `AppSKey`.

closed 2016-09-14 by htdvisser