TheThingsArchive/ttn #197: TTN no login possible via TheThingsUno (Aruino Leonardo) wich did work in Utrecht meetup 14june2016

EvertSchut, 2016-07-06:
Hi, again trying to join TTN but got: "Join not accepted" / Join not accepted: denied. Status: 00000000 / Response is not OK Join not accepted.

The Arduino Uno worked previously. I was allowed to take with me in a Meetup session june 14, Utrecht (I will share my experience) and there I got it OTAA activated and I could send "Hello World"  and Sensor data (temperature).

After only getting "response is not OK" I've modified the code and since then I can see that (1) I get an answer (response) and (2) when I get it, the response from TTN (I presume)  says "join not accepted". I conclude that the 1/2 lambda dipole antenna works here. I test in Driebergen now, but also tried Utrecht more than 1 day and more places, even mobile. So coverage shouldn't be a problem.

My TTN Dashbord on 0004A30B001B4A7E shows:
_Device type	OTAA
Status	device is activated_

In my code, I copied from Dashboard
- App EUI as lsb [8] to code: const byte appEui[8] = { (key pasted here) }
- App Key as lsb to code: const byte appKey[16] = { (key pasted here) }

So why can I not join the network? Is my device blacklisted/not allowed? Who can tell me? 

Thanks for help!

( I might go on holidays and seem less 24/7 online than some other developers, so my answer may take a while, this does not mean that I am not interested...)

**Log from my code, modified OTAA example, wich now shows the reasons of TTN:**
"Version is RN2483 1.0.1 Dec 15 2015 09:38:09, model is RN2483
Sending: mac set adr off
Sending: mac set pwridx 1
Sending: mac set dr 5
EUI: 0004A30B001B4A7E
Battery: 3263
AppEUI: 0000000000000000
DevEUI: 0004A30B001B4A7E
DevAddr: 00000000
Band: 868
Data Rate: 5
RX Delay 1: 1000
RX Delay 2: 2000

Trying to join the network
EUI: 0004A30B001B4A7E
Battery: 3263
AppEUI: 0000000000000000
DevEUI: 0004A30B001B4A7E
DevAddr: 00000000
Band: 868
Data Rate: 5
RX Delay 1: 1000
RX Delay 2: 2000
Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B4A7E
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Join not accepted: denied. Status: 00000000
..........next joining attempt..........
EUI: 0004A30B001B4A7E
Battery: 3263
AppEUI: 70B3D57ED0000405
DevEUI: 0004A30B001B4A7E
DevAddr: 00000000
Band: 868
Data Rate: 5
RX Delay 1: 1000
RX Delay 2: 2000

Trying to join the network
EUI: 0004A30B001B4A7E
Battery: 3273
AppEUI: 70B3D57ED0000405
DevEUI: 0004A30B001B4A7E
DevAddr: 00000000
Band: 868
Data Rate: 5
RX Delay 1: 1000
RX Delay 2: 2000
Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B4A7E
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Join not accepted: denied. Status: 00000000
..........next joining attempt..........
EUI: 0004A30B001B4A7E
Battery: 3273
AppEUI: 70B3D57ED0000405
DevEUI: 0004A30B001B4A7E
DevAddr: 00000000
Band: 868
Data Rate: 5
RX Delay 1: 1000
RX Delay 2: 2000

Trying to join the network
EUI: 0004A30B001B4A7E
Battery: 3263
AppEUI: 70B3D57ED0000405
DevEUI: 0004A30B001B4A7E
DevAddr: 00000000
Band: 868
Data Rate: 5
RX Delay 1: 1000
RX Delay 2: 2000
Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B4A7E
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Join not accepted: denied. Status: 00000000
..."

**Log from DHT sensor Example code:**
"Version is RN2483 1.0.1 Dec 15 2015 09:38:09, model is RN2483
Sending: mac set adr off
Sending: mac set pwridx 1
Sending: mac set dr 5
Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B4A7E
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Join not accepted
Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B4A7E
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Join not accepted
Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B4A7E
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Join not accepted
Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B4A7E
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Response is not OK
Join not accepted
"

The response is from this code:

"// Based on OTAA example code Arduino Leonardo
#include "TheThingsUno.h"

//Over the Air Activation (OTAA)
//method where devices are not initially personalized for any
//particular network but send a JOIN request to a given LoRa Network
//to receive a device address and anauthorization token from which to derive session keys. 
//
//The Application and Network session keys are derived
//during the OTA Join procedure, from a root AppKey pre-provisioned in the device as part of manufacturing or
//commissioning. OTA Activation provides a high level of security.

// Set your AppEUI and AppKey
const byte appEui[8] = {  0x70, 0xB3, 0xD5, 0x7E, 0xD0, 0x00, 0x04, 0x05  }; //for example: {0x70, 0xB3, 0xD5, 0x7E, 0xE0, 0xE0, 0x01, 0x4A1};
const byte appKey[16] = {  0xC0, 0x43, 0x9A, 0xE1, 0x9D, 0x66, 0xD4, 0x05, 0xCC, 0x12, 0x9A, 0xD7, 0x2D, 0xA4, 0xE4, 0x87 };
#define debugSerial Serial
#define loraSerial Serial1

// Set your message to send
String message = "Hello world"; //sending a string of chars "Hello world"

#define debugPrintLn(...) { if (debugSerial) debugSerial.println(__VA_ARGS__); }
#define debugPrint(...) { if (debugSerial) debugSerial.print(__VA_ARGS__); }

TheThingsUno ttu;  // make object/instance

// Requires a modified .h file on C: in which 6 rules like (readValue, sendCommand, readLine etc) 
// are made public, so the can be used here as ttu.readValue etc.
bool eigenjoin(const byte appEui[8], const byte appKey[16]) {
String devEui;
String deBuginfo = "";
  Serial.print("\nTrying to join the network\n");
    delay(2000); // this helped resolve issue sometimes not showing version and model no. 
    ttu.showStatus();
  devEui = ttu.readValue("sys get hweui");
  ttu.sendCommand("mac set appeui", appEui, 8); 
  ttu.sendCommand("mac set deveui " + devEui);
  ttu.sendCommand("mac set appkey", appKey, 16);
  ttu.sendCommand("mac join otaa");
  String response = ttu.readLine(10000);

  if (response != "accepted") {
    // Being able to display resonse this is the essentiale addition to the OTAA example, so we can now see why the join was not accepted
    debugPrintLn("Join not accepted: " + response + ". Status: " + ttu.readValue("mac get status"));
    // do not flood network but let user know we are busy (with nothing)
    int i=0;
    while ( i<10)  {  
      debugPrint(".");
      delay(5000);  
      i=i+1;  
      }
    return false;
  }
  debugPrintLn("Join accepted: " + response + ". Status: " + ttu.readValue("mac get status"));

return true;
}

// this is were the Main code starts //

void setup() {
  debugSerial.begin(115200);
  loraSerial.begin(57600);
  delay(2000);
  ttu.init(loraSerial, debugSerial);
  ttu.reset();
  delay(2000); // this helped resolve issue sometimes not showing version and model no. 
  ttu.showStatus();

  //the device will attempt a join every x miliseconds till the join is successfull
  while(!eigenjoin(appEui, appKey)){
    Serial.print("next joining attempt..........\n");
    ttu.showStatus(); // First time shows: AppEUI: 0000000000000000 ?
    // and would this be correct: DevAddr: 00000000 ?
    delay(2000); // x milliseconds
  }

  digitalWrite(13, HIGH); //turn on LED to confirm join
  delay(6000);
  ttu.showStatus();
  debugPrintLn("Setup for The Things Network complete");

  delay(1000);
}

void loop() {
  ttu.sendString(message);
      Serial.print("Sent message " + message + "\n");
  delay(3000);
}
"

This is bases on sendOTAA example, I also tried anothet example.

romeovs, 2016-07-06:
Are you within range from a gateway?

EvertSchut, 2016-07-06:
Yes, it should be, I get a response from a network. See the problem description above. When I get into a cellar, out of range, the response is not received/logged. Also tried Utrecht several places, coverage should have been there at all, diffrent chosen, places. So I conclude: I am in coverage, but the network denies the join.

htdvisser, 2016-07-06:
How do you know you are receiving a response from the network?

According to the LoRaWAN specification, the network only sends a message back when the join is accepted, and does not send a message back when the join is denied. Therefore receiving a "join denied" message would be impossible.

johanstokking, 2016-07-06:
Hi @EvertSchut, please see the [basic writing and formatting syntax](https://help.github.com/articles/basic-writing-and-formatting-syntax/) of GitHub to format your post.

When you receive "join not accepted" while your device is configured correctly, there are two possible causes: you are not in reach of a gateway, or the gateway has to be silent because of regulatory [duty cycle](https://en.wikipedia.org/wiki/Duty_cycle).

I am quite sure that coverage is the problem. Not necessarily of the gateways, but the reach of your device. You need to have an external antenna on your device. Otherwise, your reach is very limited; you should have the gateway in clear sight or on the building that you're in.

Please report back when you are physically close to a gateway.

closed 2016-07-06 by johanstokking

EvertSchut, 2016-07-06:
@johanstokking : I have an external antenna, and I visited Utrecht at diffrent days and places, where there should be coverage. I can keep on trying, but tried a lot, see initial post for details.
From the behaviour it seems really that somtimes I get a response, wich I presumed te be from the network, since in a cellar it isnt showed in the log. 
I should be close enough to a gateway.

@htdvisser : Interesting, essential point, "and does not send a message back when the join is denied", are you sure? 
How is explained that 
 debugPrintLn("Join not accepted: " + response + ". Status: " + ttu.readValue("mac get status"));
gives a parameter value of result that equals "denied"? I only get this when I seem to have coverage... Not in our cellar or with fold up antenna. 

Is there some code in the community that gives more information if there is a network present, for example RSSI/BER/Received signal indication?

htdvisser, 2016-07-06:
I am completely sure. I read the spec and I wrote the code.

The response `denied` to `mac get status` seems to be a bug in the RN2483 library.

[image]

EvertSchut, 2016-07-06:
Sorry for maybe unnecessary opening #198, it seems we can continue in this issue? Nice!

So the return value is the status of the Lora module? I havr a rn2483 indeed.

Would it be possible for the modem to tell us somthing about the received signal, or so?

johanstokking, 2016-07-06:
>>>I should be close enough to a gateway.

Which one?

Again, please report back if you are physically close to a gateway. Also please mention the concerning gateway EUI. That is the only way for us to see what's going on, because we can see the gateway activity on the network.

If you are sure that you're close to a gateway, but that there are issues with join or downlink, you can use ABP. See [example](https://github.com/TheThingsNetwork/sdk/blob/master/devices/TheThingsUno/src/examples/SendABP/SendABP.ino).

htdvisser, 2016-07-06:
As there is no message, it is not possible to get information about the received signal strength. If you would receive a message you can get the SNR with the command `radio get snr`.

I can not find any mention of your DevEUI `0004A30B001B4A7E` in the server logs, so your join attempts were definitely not received by any gateway connected to the staging environment.

EvertSchut, 2016-07-06:
@johanstokking I used your example with 3x my UI filled in.

Does the result beneath mean I am connected? 

Version is RN2483 1.0.1 Dec 15 2015 09:38:09, model is RN2483
Sending: mac set adr off
Sending: mac set pwridx 1
Sending: mac set dr 5
Sending: mac set devaddr with 4 bytes
Sending: mac set nwkskey with 16 bytes
Sending: mac set appskey with 16 bytes
Sending: mac join abp
Personalize accepted. Status: 00000001
EUI: 0004A30B001B4A7E
Battery: 3273
AppEUI: 0000000000000000
DevEUI: 0004A30B001B4A7E
DevAddr: 1CA20AB3
Band: 868
Data Rate: 5
RX Delay 1: 1000
RX Delay 2: 2000
Setup for The Things Network complete
Sending: mac tx uncnf 1 with 11 bytes
Successful transmission

On the dasboard, I do not see results.
	Dev EUI	Dev Addr	Fcnt Down	Fcnt Up
	0004A30B001B4A7E	1CA20AB3	0	166 
(It gives is 166 for weeks now, it should increase, not?)

htdvisser, 2016-07-06:
@EvertSchut: Could you please [format your code according to this guide](https://help.github.com/articles/basic-writing-and-formatting-syntax/) so that I don't have to do that every time? Thanks :)

The "Successful transmission" only means that the transmission succeeded, not that it was received by a gateway. I will check the server logs if I see anything.

EvertSchut, 2016-07-06:
Thanks, very nice that you check the logs. 
Is it possible that a am locked out, denied of service?

johanstokking, 2016-07-06:
@EvertSchut: I have updated The Things Uno SDK to version 0.13. Please [download from here](https://github.com/TheThingsNetwork/sdk/raw/master/devices/TheThingsUno/release/TheThingsUno.zip). This contains a bit more logging output for OTAA failures.

Please use the plain SendOTAA example with your `AppEUI` and `AppKey` and post the output.

Also, you cannot be locked out, denied of service or anything. You are just not close enough to a gateway, I still presume.

htdvisser, 2016-07-06:
I can not find any logs for DevAddr `1CA20AB3`, so I can conclude with 100% certainty that you are not in range of any gateway that is connected to our staging environment.

More specifically, we have not received **any** message from your device since June 14th 2016, 22:41:54

EvertSchut, 2016-07-06:
I can do a brief try near gemeentehuis Odijk, did this also before, I will do that in 15 minutes or so.

EvertSchut, 2016-07-06:
@htdvisser Okay, I'll make ure that I am in coverage.
My keys are all entered in LSB, that is okay, or not?

EvertSchut, 2016-07-06:
OK, i'll try to find coverage in 15 minutes Odijk or later in Utrecht. Thank you all very much I'll post when I did a try.

EvertSchut, 2016-07-06:
Well did you see something? I biked through Odijk and stopped near gemeentehuis were a gateway is supposed to be present.

htdvisser, 2016-07-06:
I can see logs from your device coming in between 18:38 and 18:50, so you were indeed within range of a gateway this time. Actually, messages have been received by three different gateways: `AA555A00080605B7`, `1DEE1B509773C05D` and `0000024B080E015D`.

With this I think we can consider this issue as closed, agreed?

EvertSchut, 2016-07-06:
Yes thanks, I'll try some more then, I saw the  
   Fcnt Up
    0   166  
was grown to 204 so I now know how to see the packets. Maybe the example. from johan helped, i'll try out..