TheThingsArchive/ttn #201: Is "Frames up" on TTN dashboard usful for tests, what is the update frequency?

EvertSchut, 2016-07-07:
@htdvisser  Today I drove near Uithof for coverage, and the target was to finc out wich software, or both, work. I have my modified OTAA script and ABP example wich I got yesterday, something worked according to your loggings.

So to find out which script did the trick, I drove around with my Arduimo LoraOne om a suction holder outside my car window and tried both scipts. I was checking Frames up	435 / Fcnt Up 435 and expected this number to grow as I got connection. After the tests we ditd yesterdag, it was grown from 204 to 435. But it is still at 435, so it seems I threw away 2 hours of drive tests. Or is this not a smart method? Is the Fcnt Up up only updataed every 12/24 hours or so? 

What is the "Frames Up" of "Fcnt Up" update-interval?
And is this a good parameter/indication  to check wether  you are reaching TTN or not?

EvertSchut, 2016-07-07:
I also drove to Odijk were you saw packets come in, from 13:58 I used ABB script, after 14:00 I used OTAA example. Did you see packets, when?

htdvisser, 2016-07-07:
I'm sorry, but I don't have the time to go through the server logs every time to check if your node is working.

The dashboard receives live updates when the frame counter increases and updates it immediately. I suggest you read [this information about frame counters](http://staging.thethingsnetwork.org/wiki/LoRaWAN/Security#security-in-lorawan-and-ttn_frame-counters), as it explains why your counters are not increasing.

Also I think it's better to ask these kind of questions on [Slack](http://slack.thethingsnetwork.org/) instead of creating issues here on Github. This way, we can use Github for real issues that are related to the implementation of The Things Network.

closed 2016-07-07 by htdvisser