TheThingsArchive/ttn #203: Disable persistent sessions in MQTT

johanstokking, 2016-07-08:
Currently, the MQTT broker allows persistent clients (clients with clean sessions disabled), e.g. they rely on the broker to restore the session with subscriptions. This is not safe as the MQTT broker always needs to remain state of the clients and subscriptions, and also share this state among brokers when we scale it horizontally. When the MQTT broker is unable to restore the session, the MQTT protocol does not provide a way to inform the client. Disconnecting the client is not an option, because of auto reconnect, the client will keep connecting automatically.

- [ ] Do not allow clients with clean session disabled on the MQTT broker *
- [ ] The MQTT broker should not send PUBACK or SUBACK when QoS 2 is requested *
- [ ] Update the default MQTT client to enable clean sessions (as part of #199)
- [ ] Update the SDKs and examples to enable clean sessions

Note: this breaks QoS 1 and 2 for offline clients. QoS 1 is still supported when the client is online.

* this may require forking Mosquitto. We may need to consider building our own MQTT broker, which becomes way easier if we don't need to support persistent sessions.

kgbvax, 2016-08-04:
I think that we will loose too much if we throw all this overboard. 
Where is the point of having a queue when it is not queuing anything? (At least this is how I understand this issue).

Being able to have persitent session and "work" in the eye of failing clients and connections is one of the major benefits of the async model.

I don't get what the problem with mosquitto is but perhaps you want to look at emqttd?
I ran a public broker for a while and persistent session were not a problem as the broker has a high-water mark for the number of messages and duration how long it would retain messages for a single persistent client, preventing it from overflowing.

johanstokking, 2016-08-04:
Although the Q in MQTT stands for Queuing, it's just a feature (and an old name). MQTT is primarily a publish/subscribe protocol, and that is what we need.

The problem is not so much in the broker, but more in the behavior of the clients when they are persistent.

When clients are persistent, they require the broker to restore the session on reconnect, so they subscribe once on start-up (which may be months ago). That is by design of MQTT. Now, if the broker is unable to restore the session, because of expiration or losing state, it has no way to inform the clients, so they are connected but they are not subscribed, let alone that they will get their lost messages.

If queuing is really necessary, then we might need to use AMQP for that. MQTT is just not the way to go here in this scenario.

kgbvax, 2016-08-04:
>it has no way to inform the clients, so they are connected but they are not subscribed, let alone that they will get their lost messages.

In MQTT 3.1.1 this is solved via the "Session Present" flag, isn't it?

johanstokking, 2016-08-04:
It is, yes, but it is not widely supported by client libraries. If you can point me to one that reads the flag in the first place, please, and then it has to handle it correctly too, like subscribing on reconnect if the session is not there.

On paper it works. In practice, we're gonna lose session state, it's gonna be hard to replicate between brokers if we scale and users will subscribe once after connect, as in almost all code samples, so their smart MQTT client would have to restore it. That's my concern.

htdvisser, 2016-10-25:
For now we are not disabling persistent sessions in MQTT. With the AMQP interface being added to the Handler, most issues can be easily solved by using that.

commit 4c84ae00 (2016-07-19): Update MQTT Client for #203

- Change QoS to 0x00
- Set Clean Session to false
- Keep track of subscriptions
- Re-subscribe on reconnect
- Add Unsubscribe funcs

Closes #203
closed 2016-10-25 by htdvisser