TheThingsArchive/ttn #205: https://github.com/ttn-zh/ic880a-gateway.git/ not found

skramer, 2016-07-09:
git clone https://github.com/ttn-zh/ic880a-gateway.git ~/ic880a-gateway
after entering github user/pass

and I get back a message:
fatal: repository git clone https://github.com/ttn-zh/ic880a-gateway.git/ not found
(preventing us from completing our The Things Network LoRa Gateway.. sad... )

The above command comes from:
https://github.com/ttn-zh/ic880a-gateway
section: Clone the installer and start the installation

Can anyone advise what the issue is.. 
Thank you in advance

johanstokking, 2016-07-09:
This issue is not related to this repository. I see that you already filed the issue at [the right place](https://github.com/ttn-zh/ic880a-gateway/issues/8).

closed 2016-07-09 by johanstokking

skramer, 2016-07-09:
thank you kindly .. I will await their reply...