TheThingsArchive/ttn #207: Dashboard: QR code per device

jpmeijers, 2016-07-21:
It would be nice if the dashboard would provide a QR code under each device. The QR code should contain the information necessary to obtain the node's data via MQTT.

For TTN Mapper we are currently using this format in the iOS app: [appEUI]:[devEUI]:[access key]
At the moment we encourage people to create their own QR code with an online tool, and then scan it with the app so that all the codes are automatically filled in.

Having this option will save people a lot of effort to type the keys over by hand. iOS also does not allow the copy-paste option from the browser to the app, like in Android.

Example:
[image]

romeovs, 2016-07-22:
I like the idea!
Do you think this is applicable to other use cases as well?

jpmeijers, 2016-07-22:
Personally I think this is the type of feature that will get more use cases after it is implemented.

I imagine buying a sensor node from someone, scanning the QR code printed on the enclosure with the manufacturer's app, and then everything is automatically linked and data is displayed in-app.

johanstokking, 2016-07-25:
I like the idea.

I see two different use cases here: 1) getting data from a device for authorized users (e.g. logged in in the dashboard), and 2) to register devices easily.

Regarding 1): for this, we'd actually need a device access key. The application access key gives too much access. Still, there may be a lot of use cases for this.

Regarding 2): I think what's useful is that devices configured for OTAA have a QR code on it with the AppEUI, DevEUI and AppKey, and devices configured for ABP a QR code with DevAddr, AppSKey and NwkSKey. A The Things Network app would make it super easy to register these devices for your application. Note that we're allowing any AppEUI for your application in the near future.

What do you think?

romeovs, 2017-05-23:
Do we still want/need this? Otherwise we can close the issue.

jpmeijers, 2017-05-23:
I still think this is a nice feature to "link" a device to a mobile app. It however does not have priority at the moment.

closed 2017-05-23 by johanstokking