TheThingsArchive/ttn #210: Adapt ttnctl to XDG base directory specification

b1101, 2016-08-02:
In the current implementation the `ttnctl` config path is hardcoded to be searched for in `$HOME` (in [ttnctl/util/config.go](https://github.com/TheThingsNetwork/ttn/blob/242ec0cb880a9086836a8dda55de622044717ff8/ttnctl/util/config.go#L18-L32) and in [ttnctl/cmd/root.go](https://github.com/TheThingsNetwork/ttn/blob/242ec0cb880a9086836a8dda55de622044717ff8/ttnctl/cmd/root.go#L104-L108)

In Linux environments it make sense to first search in `$XDG_CONFIG_HOME/ttn`, if the variable is set. See [XDG base directory specfication](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html)

In the current implementation it would be necessary to make a special case, as the config file name would have to not be a dotfile(as in `.ttnctl.yaml`)

romeovs, 2016-09-22:
I will implement this when implementing the account server client token manager.

Having one token in the config is ok-ish, but having multiple tokens will be a pain. Instead, tokens should be stored in `$XDG_CACHE_HOME/ttnctl/tokens` or something similar.

htdvisser, 2016-09-28:
Also create the config file if it doesn't exist (to avoid warning/error)

htdvisser, 2016-10-03:
@romeovs implemented this last week.

PR #282 (2016-09-30): Feature/xdg
Fixes #210.

The location of the config file is now looked up in this order:
1. `--config` flag
2.`$XDG_CONFIG_HOME/ttnctl/config.yml`
3. `$HOME/.ttnctl.yml`

Similarly, the directory for storing data is inferred as follows:
1. `--data` flag
2.`$XDG_DATA_HOME/ttnctl`
3.`$XDG_CACHE_HOME/ttnctl`
4. `$HOME/.ttnctl`

Furthermore, `ttnctl` no longer writes data to the config file (for instance to store app ID an EUI preferences), but writes preferences to the data directory.
closed 2016-10-03 by htdvisser