TheThingsArchive/ttn #221: [dashboard] 500 when using a invalid jwt

cambierr, 2016-08-16:
I noticed that providing a valid, but out-of-scope jwt to the server creates a 500.
It happens about 1 times on 5 when creating applications.
It comes from the fact that sometimes, the dashboard does not force-refresh the jwt

cambierr, 2016-08-16:
ping @romeovs ?

romeovs, 2016-08-16:
Thanks for the report!
Can you see the error message by any chance?

cambierr, 2016-08-16:
here it is: {"reason":"rpc error: code = 2 desc = Invalid operation: rpc error: code = 2 desc = Invalid operation: Unauthorized"}

Btw, the app_eui sent in the json body seems to be ignored and only the one in the url is taken in account.

johanstokking, 2016-08-16:
The API is for the dashboard and is (currently) not open source, documented and supported. Breaking changes can be deployed any time.

Please use the gRPC services (see https://github.com/TheThingsNetwork/ttn/tree/refactor/api) to interface with the handler. We're working on documentation for the account server API, but that's also still work in progress.

Note that application management is part of the account server. There is no database behind the dashboard API; it only interfaces with the handler, MQTT broker and account server.

cambierr, 2016-08-16:
I know that (we already use it) but this one is an issue we have sometimes in the dashboard

johanstokking, 2016-08-16:
Ah, misunderstood, sorry about that. @romeovs will take this. Thanks for reporting.

htdvisser, 2016-10-03:
Fix deployed in preview dashboard

closed 2016-10-03 by htdvisser