TheThingsArchive/ttn #230: grpc: Use google.protobuf.Empty instead of api.Ack

rvolosatovs, 2016-09-07:
The protobuf stdlib provides an [Empty](https://developers.google.com/protocol-buffers/docs/reference/google.protobuf#google.protobuf.Empty) type, which could be used instead of custom api.Ack type

AntoineRondelet, 2016-09-07:
Don't forget to do

`go get -v github.com/golang/protobuf/ptypes`
and  `import "github.com/golang/protobuf/ptypes/empty/empty.proto";` in your proto
If you want to use it

johanstokking, 2016-09-07:
@AntoineRondelet This is builtin no? Looks like it's in the protobuf src

htdvisser, 2016-09-07:
Done for `ttn@refactor`.

htdvisser, 2016-09-08:
Also, you should not need to import the `.proto` like @AntoineRondelet commented. If your protobuf installation is correct, these protos should be included in the compiler, so you can just use `import "google/protobuf/empty.proto";`.

htdvisser, 2016-09-13:
Done

commit 37f54d1d (2016-09-07): Use Empty type instead of api.Ack

Resolves #230 for ttn part
closed 2016-09-13 by htdvisser