TheThingsArchive/ttn #234: MQTT: Failed activations are also published

FokkeZB, 2016-09-07:
I noticed while updating the Node-RED node that failed activations are also published:

Arduino Serial Monitor logs:

```
Start setup
Version is RN2483 1.0.1 Dec 15 2015 09:38:09, model is RN2483
Sending: mac set adr off
Sending: mac set pwridx 1
Sending: mac set dr 5
Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B7AD2
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Join not accepted: denied

Sending: mac set appeui with 8 bytes
Sending: mac set deveui 0004A30B001B7AD2
Sending: mac set appkey with 16 bytes
Sending: mac join otaa
Join accepted. Status: 00000401
EUI: 0004A30B001B7AD2
Battery: 3223AppEUI: 70B3D57ED0000AFB
DevEUI: 0004A30B001B7AD2
DevAddr: 26012036
Band: 868
Data Rate: 5
RX Delay 1: 1000
RX Delay 2: 2000
Setup for The Things Network complete

Sending: mac tx uncnf 1 with 5 bytes
Successful transmission
```

Node-RED debug logs:

```
9/7/2016, 5:01:52 PMe0c21758.740398
msg.payload : Object
{ "dev_id": "FZB-UNO", "app_id": "FZB-HELLO", "app_eui": "70B3D57ED0000AFB", "dev_eui": "0004A30B001B7AD2", "dev_addr": "26012D97", "metadata": { "time": "2016-09-07T15:01:52.64594287Z", "frequency": 867.9, "modulation": "LORA", "data_rate": "SF7BW125", "coding_rate": "4/5", "gateways": [ { "eui": "0000024B08060112", "timestamp": 3271049259, "time": "2016-09-07T15:01:52.609908Z", "channel": 2, "rssi": -107, "snr": 1.2 } ] } }

9/7/2016, 5:02:05 PMe0c21758.740398
msg.payload : Object
{ "dev_id": "FZB-UNO", "app_id": "FZB-HELLO", "app_eui": "70B3D57ED0000AFB", "dev_eui": "0004A30B001B7AD2", "dev_addr": "26012036", "metadata": { "time": "2016-09-07T15:02:05.013990835Z", "frequency": 868.1, "modulation": "LORA", "data_rate": "SF7BW125", "coding_rate": "4/5", "gateways": [ { "eui": "0000024B08060112", "timestamp": 3283418731, "time": "2016-09-07T15:02:04.979378Z", "rssi": -109, "snr": 6, "rf_chain": 1 } ] } }

9/7/2016, 5:02:10 PMe0c21758.740398
msg.payload : Object
{ "message": "Hello" }
```

htdvisser, 2016-09-07:
Joins are published only when they are accepted by the backend. If the node does not receive the accept message (because it gets lost somewhere), the node will think the join was denied, even when it was accepted.

Unfortunate situation.

closed 2016-09-07 by htdvisser

FokkeZB, 2016-09-07:
@htdvisser but can't the backend see the device already activated (verify keys still work somehow) and not give out new session nor broadcast new activation?

htdvisser, 2016-09-08:
No, an activation request from the device indicates that it wants to start a new session.

FokkeZB, 2016-09-08:
Mmm, yes.. there's no way it could tell the difference of course unless we do another uplink to confirm receiving the confirmation.. haha.

OK, I will add a note in the guides somewhere.

htdvisser, 2016-09-08:
For the documentation:

A "denied" join can have several reasons:

1. Join-Request not received by server - can be caused by message loss in the wireless communication (out of range)
2. Join-Request not accepted by server - device not known, incorrect signature (MIC)
3. Join-Accept not received by node - can be caused by message loss in the wireless communication (out of range)

When an activation is received on MQTT, it means that at least 1 and 2 were not the case. The third might still happen.

The device should just retry a couple of times (possibly with exponential backoff).

FokkeZB, 2016-09-08:
Great info, thx!