TheThingsArchive/ttn #244: Add more data monitoring to gateways

cambierr, 2016-09-15:
It could be great to add more monitoring fields to gateways.
In our case, we are monitoring cpu load, ram % use, bandwidth, packet losses, and two temperatures.

This could be achieved by adding a repeated meta field which is a Key/Value pair for instance, so that custom monitoring is possible by adding fields in different gateways / forwarder implementations

sieben, 2016-10-04:
Would be good to add stats on:

- [ ] signal strengths and signal/noise ratios too
- [ ] activity 
- [ ] rx/tx counters

htdvisser, 2016-10-20:
We already have the following optional fields in `stat` messages:

- `"pfrm":"Kerlink"` - the gateway platform
- `"mail":"contact@email.network"` - contact info of the gateway owner
- `"desc":"Contacts Home Gateway"` - description of the gateway

It would be possible to add some system metrics to the packet_forwarder protocol like @cambierr suggests. We'd have to specify a set of "supported" fields (because sending arbitrary data is usually a bad idea):

- `"load":"1.54 1.93 2.03"` - system load of the gateway host 
- `"cpu":12` - average total CPU usage (in percent) since the last `stat` message
- `"mem":12` - total memory usage (in percent)
- `"temp":31.5` - temperature

The other statistics mentioned, can easily be determined on the server-side.

- RSSI/SNR is added to each uplink message
- Rx/Tx is already in the packet_forwarder protocol
- Activity can easily be measured by looking at the Uplink/Downlink that pass through the backend
- Bandwidth (at least to the backend) can be calculated by the backend

---

Implementing the system metrics fields is not much work on the server side, and it sounds like @cambierr already implemented this for the packet forwarder. Let's try to formalize a specification for these fields (also the existing ones, because those are not really documented either).

closed 2016-11-30 by htdvisser

htdvisser, 2016-11-30:
The `gateway.Status` proto now contains the following fields:

```
float   load_1
float   load_5
float   load_15
float   cpu_percentage
float   memory_percentage
float   temperature
```

These fields will have to be filled by the gateway. If the gateway uses the `ttn-gateway-connector` this can be done by simply filling the protocol buffer. The packet forwarder does not support these fields.

commit 249b772a (2016-11-30): Add host system metrics to gateway status proto

Closes #244