TheThingsArchive/ttn #247: Payload Functions: Pass port as second argument

FokkeZB, 2016-09-16:
Once this is in place here, https://github.com/TheThingsIndustries/dashboard/issues/74 requests for the placeholders to demonstrate usage of the port.

Users will address different types of messages to different ports. In the payload functions I need to know to what port the message was addressed. Currently, I don't have this information.

I'd suggest we add the port as second argument to all 4 functions:

```js
function Decoder(bytes, port) {
  // Decode an uplink message from
  // a Buffer of bytes to an object.

  decodedObj = {};

  if (port === 1) {
    decodedObj.isLightOn = bytes[0];
  }

  return decodedObj;
}
```

htdvisser, 2016-09-24:
@johanstokking promised on Slack to also add frame counters to the payload functions (https://thethingsnetwork.slack.com/archives/backend/p1474722721000621). I don't see the added value of adding these counters, especially because they are only relevant for LoRaWAN devices. Could you please elaborate, @johanstokking?

johanstokking, 2016-09-26:
Yeah we don't need the frame counters indeed, but port is essential.

FokkeZB, 2016-10-10:
Can we get this port thing rolling? I need it for the Electra demo and Node QS

romeovs, 2016-10-25:
Closed by #329

closed 2016-10-25 by romeovs

FokkeZB, 2016-10-25:
@romeovs could you create an issue or PR in the docs repo when you make a change that affects users?

https://github.com/TheThingsNetwork/docs/issues/66

FokkeZB, 2016-11-01:
Shouldn't these be updated as well?

- https://github.com/TheThingsNetwork/ttn/blob/v2-preview/core/handler/convert_fields_test.go#L50
- https://github.com/TheThingsNetwork/ttn/blob/v2-preview/core/handler/dry_run_test.go#L77