TheThingsArchive/ttn #249: Payload Functions: Combine Decoder/Converter/Validator

FokkeZB, 2016-09-16:
For uplink we have a separate Decoder, Converter and Validator. For downlink we only have an Encoder, although #248 discusses it's perfectly capable of dropping messages.

I'd suggest we drop Converter and Validator. You can do whatever you do in Converter in Decoder and to drop you'd simply return false, just like #248 suggests for the Encoder.

This would make it more clear that one function is for down and the other for up, while now you might think the Converter and Validator act on both ways.

htdvisser, 2016-09-16:
I like it. @johanstokking?

johanstokking, 2016-09-16:
I like separation of concerns. Also, I like the ability to replace a JavaScript decoder by a Google Protocol Buffers decoder, and keep the converter and validator in place.

FokkeZB, 2016-09-16:
So why don't we have that separated for downlink as well?

Can we at least group the first 3 so that it is clear that those are for up and encoder for down?

johanstokking, 2016-09-17:
Good to group. See #248 for the other issue. Summary; de node can be stupid; send analog readings without any conversions or validations to save power. An application doesn't need that.

closed 2016-09-17 by johanstokking