TheThingsArchive/ttn #252: MQTT: Publish/Accept payload_raw as array of numbers

FokkeZB, 2016-09-19:
We publish and accept (via downlink) `payload_raw` base64 encoded.

Earlier, we also displayed the (raw) payload base64 encoded on the dashboard, but [changed this to hex](https://github.com/TheThingsIndustries/dashboard/issues/49) because it's easier to read, to recognise as that what you've send (since you probably wrote and send it in hex as well) and saves novice users another encoding/decoding to learn.

For the same reasons I suggest we also get rid of base64 for the `up/down` `payload_raw` field of the MQTT broker and instead use an array of numbers. This again is more readable and also easier to process.

Yes, it would make message a bit larger, but without the payloads the message is already over 300 characters and with most payloads this won't really be that big of a difference.

**base64 encoded** (89)

```
/////////////////////////////////////////////////////////////////////////////////////w==
```

**array of numbers** (258)

```
[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255]
```

htdvisser, 2016-09-19:
👎 

1. The argument that hex is easier to read than base64 does not really justify moving from base64 to an array of bytes.
2. We have to make clear that the `payload_raw` field contains binary data. When we use base64 this should be clear to most developers. The suggested array of bytes requires us to explain to people that the numbers should be in the 0-255 range (and we'd have to return some kind of error when they don't do this, which is quite difficult with MQTT)
3. We're building great SDKs, so for people who don't use those, we can assume they have enough technical knowledge to convert from base64 to hex.

johanstokking, 2016-09-19:
Yes, agreed with 1-3.

closed 2016-09-19 by johanstokking

FokkeZB, 2016-09-19:
I'm just asking stupid questions before our users do ;) Thanks for the excellent answers.

Would it be helpful for our app libraries to feature either:

- A method to decode a base64 string to a Buffer
- To decode `payload_raw` to a Buffer before passing it on? (my fav)

johanstokking, 2016-09-19:
Yes, but that's really language-dependent. We should make it available as the native binary buffer, so in Node.js that's `Buffer`, in Go `[]byte`, etc. This may go automatically if we're using a typed deserializer.

htdvisser, 2016-09-19:
👍 I think the SDKs should indeed convert `payload_raw` into a native binary buffer.

glangaas, 2018-01-31:
Hello, We receive the payload similar to this (payload_raw):  ..."is_retry":true,"payload_raw":"AmWwPHvRAA==","metadata":{"time":"...  On the things server the data looks like 02B7A53C46D200 -   How will the Payload Format become if we just want to send data straight thru?   Or how do we convert the payload_raw data such it is in hex again?  

Thanks,
Gaute

jpmeijers, 2018-02-05:
@glangaas the `payload_raw` is base64 encoded. When you use a converting utility like http://www.asciitohex.com/ you can see `AmWwPHvRAA==` translates to `02 65 b0 3c 7b d1 00` which is the format in which you sent it. Please try and keep this type of support questions to the forum or Slack.