TheThingsArchive/ttn #254: Report payload function errors back over MQTT

johanstokking, 2016-09-19:
If a payload function fails, the processing is stopped and there should be a way to inform the user (see conversation in #248).

The extent to which we make these errors available to the end user, e.g. in the dashboard or ttnctl, is to define later, but it all starts with reporting back on MQTT.

For uplink payload function errors, I propose reporting the error to `:app-id/devices/:dev-id/up/rejected`, and for downlink `:app-id/devices/:dev-id/down/rejected`. This follows [AWS IoT naming](http://docs.aws.amazon.com/iot/latest/developerguide/thing-shadow-mqtt.html), and allows for a companion named `accepted`.

The error should indicate the type of error (payload function) and the concerning payload function (decoder, converter or validator).

The type of error should be clear, because there are other use cases for this topic. For example, we can use the `.../up/rejected` also for payload marked as invalid by the validator, and `.../down/rejected` in cases where we can't find a gateway for downlink in time.

htdvisser, 2016-09-19:
Related to https://github.com/TheThingsNetwork/ttn/issues/159

johanstokking, 2016-09-19:
The question here is if we should put something behind `:app-id/devices/:dev-id/up/...`, because we're also publishing payload fields here. So `accepted` and `rejected` become reserved words for fields.

johanstokking, 2016-09-19:
OK, so after a bit of internal discussion regarding naming, reserved words, broader use as in #159, an application events topic is a better option.

Errors would be reported to `:app-id/events/errors`, other information may fall also under `events`. Monitoring can listen to `:app-id/events/#`. All messages posted in that tree should have the same payload format.

htdvisser, 2016-09-22:
Closed with https://github.com/TheThingsNetwork/ttn/issues/159

Errors are reported in `:app-id/devices/:dev-id/events/up/errors` or `:app-id/devices/:dev-id/events/down/errors`

closed 2016-09-22 by htdvisser