TheThingsArchive/ttn #255: Add ttnctl to package managers

johanstokking, 2016-09-19:
- [ ] [Brew](https://github.com/Homebrew/brew/blob/master/share/doc/homebrew/Formula-Cookbook.md)
- [ ] Apt repository

dorianamouroux, 2016-09-19:
This one can be considered as well http://snapcraft.io/

htdvisser, 2016-09-19:
Already looked into Homebrew. We need to set up our own Tap for this, which is not quite hard.

FokkeZB, 2016-09-19:
@dorianamouroux snap craft looks very polished. I'll look into all of them this week.

FokkeZB, 2016-09-26:
Brew doc was moved: https://github.com/Homebrew/brew/blob/master/docs/Formula-Cookbook.md

FokkeZB, 2016-09-26:
I can do https://github.com/Homebrew/brew/blob/master/docs/Formula-Cookbook.md but it might make more sense for @htdvisser or someone else who's working on `ttnctl` to do this?

FokkeZB, 2016-10-06:
@htdvisser could you self assign this issue? I can't, but since you're already on it.

joscha, 2017-03-27:
references https://github.com/Homebrew/homebrew-core/pull/10918

htdvisser, 2017-06-22:
Homebrew formula is now done. Currently in a [separate tap](https://github.com/TheThingsNetwork/homebrew-ttn) but we might want to submit a PR for homebrew-core when we're sure the builds are good.

In the meantime, this is what you need:

```
brew tap TheThingsNetwork/ttn
brew install ttnctl
```

dbrgn, 2017-07-07:
Here's an Arch Linux AUR package: https://aur.archlinux.org/packages/ttnctl-bin/

htdvisser, 2017-07-09:
Great work! Do you think we can somehow automate updating the package when new versions of ttnctl are released?

dbrgn, 2017-07-09:
@htdvisser people can flag the package as out-of date on the AUR, then I get a notification. that mechanism usually works well. alternatively an e-mail notification by TTN on new releases for packagers would be nice too.

htdvisser, 2017-07-09:
Will you be able to maintain the package, otherwise it might be a good idea to publish the package from a TTN account?

You can subscribe to new releases using feed:https://github.com/TheThingsNetwork/ttn/releases.atom

dbrgn, 2017-07-09:
Sure, you can take over if you want. Do you have someone who is familiar with the PKGBUILD system?

As for the build script itself, it's [pretty simple](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ttnctl-bin).

htdvisser, 2017-07-10:
@rvolosatovs maybe?

rvolosatovs, 2017-07-10:
I could do that.
Why don't we also set up a `ttnctl-git`, which would build from `master`, such that users always use up-to-date version? I think many arch users would prefer that over binaries and it almost doesn't need maintenance.
On the side note, I also propose adding a package in https://github.com/NixOS/nixpkgs , which I can maintain.

dbrgn, 2017-07-10:
Sure, a `ttnctl-git` would also be an option.

@rvolosatovs what's your AUR username? I think I can add you as a contributor.

htdvisser, 2017-07-10:
I can imagine that many people just want to install `ttnctl`, not install Go+all vendors and spend several minutes building from source, but sure, we could add a git/`master` alternative if needed. Should not replace the bin of course

rvolosatovs, 2017-07-10:
@dbrgn rvolosatovs
@htdvisser I will

dbrgn, 2017-07-10:
@rvolosatovs added you as maintainer. I think you should be able to publish releases too now.

htdvisser, 2019-02-18:
We will no longer do this for v2, but it will be done for v3: https://github.com/TheThingsNetwork/homebrew-lorawan-stack

closed 2019-02-18 by htdvisser